OnionUI/Onion #846: ROMs containing ´ in the name cannot be deleted from select menu

ricsim, 2023-05-29:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2 beta

### Miyoo Firmware Version

20230329

### Provide a clear and concise description of the issue

When a ROM contains ´ in the name, it cannot be deleted from the select menu.

### Steps to Reproduce

Try to delete eg this ROM
Roger Clemens' MVP Baseball (USA) - Copy.zip
it will not delete it due to the character after Clemens

### Provide any additional context or information that may help us investigate your issue

Delete fails

ricsim, 2023-05-29:
It seems that in the latest beta it works.
But it does not delete the file from the cached ROM list.
In previous OnionOS versions it was deleted from the list (except if it contained a ´) now it seems to be gone from the list after a refresh (but even IF it contains a ´).

Aemiii91, 2023-05-29:
This feature is part of MainUI (Miyoo's closed source frontend)

closed 2023-05-29 by Aemiii91