OnionUI/Onion #851: RTC (clock) - "Set automatically from network" not working.

duckyb, 2023-05-30:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion V4.2.0 beta 4

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

Despite the new setting being activated, the clock does not update when connecting to wifi.

[image]

### Steps to Reproduce

1. Activate "tweaks > system > date & time > **set automatically from network**
2. Turn wifi OFF
3. Turn wifi back ON
4. Click on network to connect
5. Open "Clock" app to check the set time
6. The time is incorrect

### Provide any additional context or information that may help us investigate your issue

Other wifi functions like FTP work properly

XK9274, 2023-05-30:
Hey,

After running:

```Activate "tweaks > system > date & time > set automatically from network
Turn wifi OFF
Turn wifi back ON
Click on network to connect
Open "Clock" app to check the set time
The time is incorrect
```

Can you check the contents of /mnt/SDCARD/.tmp_update/logs/network.log please?

If you're starting NTP before enabling WIFI then NTP won't run that cycle and it'll be disabled, this is by design so there isn't processes running in the background (taking up resources) that aren't able to contribute anything due to lack of internet access. All the other network processes will behave the same and you'll generally see:

```
NTP: Wifi is turned off, disabling the toggle for NTP and killing the process
Hotspot: Wifi is turned off, disabling the toggle for hotspot and killing the process
Filebrowser: Wifi is turned off, disabling the toggle for HTTP FS and killing the process
Telnet: Wifi is turned off, disabling the toggle for Telnet and killing the process
Dropbear: Wifi is turned off, disabling the toggle for dropbear and killing the process
FTP: Wifi is turned off, disabling the toggle for FTP and killing the process
```

duckyb, 2023-05-30:
```
Mon May 29 18:29:24 UTC 2023 Network Checker: Update networking
Mon May 29 18:29:24 UTC 2023 Network Checker: Initializing wifi...
Mon May 29 20:29:27 UTC 2023 FTP: Starting bftpd
Mon May 29 20:29:27 UTC 2023 Telnet: Killed
Mon May 29 20:29:27 UTC 2023 NTP: Starting NTP with TZ of UTC-2
Mon May 29 20:29:27 UTC 2023 NTP: TZ set to UTC-2, Time set to: Mon May 29 20:29:27 UTC 2023 and merged to hwclock, which shows: Mon May 29 20:29:28 2023 0.000000 seconds
```

This is what I got. It seems like it updated the time successfully but the time set is incorrect.
UTC+2 time should be `Tue May 30 17:29:27 UTC 2023`

XK9274, 2023-05-30:
That's odd, as setting mine to the same gives (the +/- is incorrect here in the logs but that's a known issue):

```
Tue May 30 00:13:30 UTC 2023 Network Checker: Update networking
Tue May 30 00:13:30 UTC 2023 Network Checker: Initializing wifi...
Tue May 30 01:13:33 UTC 2023 FTP: Starting bftpd
Tue May 30 01:13:33 UTC 2023 Dropbear: Starting dropbear
Tue May 30 01:13:33 UTC 2023 Filebrowser: Starting filebrowser listening on 0.0.0.0 to accept all traffic
Tue May 30 00:16:14 UTC 2023 Network Checker: Update networking
Tue May 30 17:41:02 UTC 2023 NTP: Starting NTP with TZ of UTC-2
Tue May 30 17:41:02 UTC 2023 NTP: TZ set to UTC-2, Time set to: Tue May 30 17:41:02 UTC 2023 and merged to hwclock, which shows: Tue May 30 17:41:03 2023 0.000000 seconds
```

Can you try the following in a Putty/hyper terminal:

export TZ=UTC+2
ntpd -p time.google.com
hwclock -w
date

The problem i had when i was looking at NTP was that ntpd seems to accept POSIX tz but not zone ID (Europe/Paris) so it looks like the offset is broken. UTC may not be the best way to handle this. I'll look into it further soon.

duckyb, 2023-05-30:
[image]

XK9274, 2023-05-30:
```
export TZ=UTC-2
ntpd -p time.google.com
hwclock -w
date
hwclock
should show correct for you, but i adjusted for this in the NTP scripting
```

Adding this here for other peoples benefit. 

When scripting for NTPD the timezones seemed to be backwards.. If you set UTC+2 you'll get UTC-2 and vice versa.

The issue is that we moved some ntpd config out over runtime.sh into another script that now means ntpd runs but it runs in the new shell environment where the TZ isn't set (FTP file times are probably correct). Setting the TZ in whatever env you need should fix this... looking at what can be changed now to fix across all envs....

XK9274, 2023-05-30:
https://github.com/OnionUI/Onion/pull/853

PR created that should address the main problems with this.

duckyb, 2023-06-01:
Beta 5 works as intended. Thank you for the fix!

closed 2023-06-01 by duckyb

XK9274, 2023-06-02:
Great, thanks for letting us know :)