OnionUI/Onion #854: Exiting Activity Tracker causes crash. Error code 139

iFiNi22, 2023-05-31:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Latest 4.2.0 Beta 4

### Miyoo Firmware Version

2023032622339

### Provide a clear and concise description of the issue

Exiting Activity Tracker causes crash. Error code 139

### Steps to Reproduce

Enter Activity Tracker, then exit by pressing B. Fatal Error code 139 appears then crash out

### Provide any additional context or information that may help us investigate your issue

[image]

Ayosoap, 2023-06-01:
I got this problem too.

Aemiii91, 2023-07-04:
Fixed in #956