OnionUI/Onion #862: retroarch settings don’t save

puggersbepugging, 2023-06-01:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

420 beta 4

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

when i go into retroarch settings and turn off auto save state and load state automatically, it always turns back on no matter what

### Steps to Reproduce

turn off the settings, exit retroarch, reenter retroarch

### Provide any additional context or information that may help us investigate your issue

i wanted to turn this off cuz every time i boot a rom, it boots the save stage but i want it to be like a fresh start. so i turned off auto load save and whatnot and it just doesn’t stick

Aemiii91, 2023-06-01:
Not a bug. You have to save the configuration - see RGC's [RA Starter Guide](https://retrogamecorps.com/2022/02/28/retroarch-starter-guide/).

closed 2023-06-01 by Aemiii91

puggersbepugging, 2023-06-01:
oops, sorry about that… thanks for all your work