OnionUI/Onion #866: Brightness level

Amiga500, 2023-06-01:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion-v4.2.0.3-beta-dev-d937d44

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

If i change brightness into settings with cursor i can low it until 01/10, but if i try to low it with MENU+VOLDOWN it decrease until 00/10. It's normal this behaviour? 

### Steps to Reproduce

Go into Settings -> Brightness and use the left cursor to minimize the brightness. it will reach 01/10
Now try with MENU+VOLDOWN and it will reach 00/10

### Provide any additional context or information that may help us investigate your issue

*

Aemiii91, 2023-06-01:
This is not the place to ask such questions. And yes, it is intended behaviour. MainUI is not made by us.

closed 2023-06-01 by Aemiii91

Amiga500, 2023-06-01:
ok, thanks and sorry