OnionUI/Onion #870: [MainUI] - Feature request : Detailed Games info

schmurtzm, 2023-06-02:
Now that scraping is possible from the MM+ it could be nice to have more detailed info on games :

[image]

kj20029, 2023-06-21:
Ilove the idea to scraping art and info but Wouldnt it conflict with the media that you add from other means like I personally am using Skraper as my Wifi is rubbish and I use a certain mix i got online that has  different layout then the one you get from Scrapring from onion os directly would it replace all the art work also ? 
FYI the mix im using is by AchillesPDX called Big zoom 
[image]

schmurtzm, 2023-06-21:
There is already a file called miyoogamelist.xml which is in the same format as the gamelist.xml generated by Skraper. The idea is to parse this file to display it in MainUI.

kj20029, 2023-06-23:
Then that would just be really great if it could work