OnionUI/Onion #872: Wiki Update Title

scottchiefbaker, 2023-06-02:
### Checklist

- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and something is incorrect or missing

### Wiki Section

FAQ

### Description

The HTTP servers and the SSH login are both amazing features of OnionOS. There is no documentation that I could find about what the credentials for those services are. There doesn't appear to be a way in the OS to set a password either. The default username and password should be documented as a FAQ somewhere.

### Correction

Include the credentials for SSH and HTTP server in the FAQ. Even better would be instructions for how to set those passwords.

Aemiii91, 2023-06-02:
We are in the process of documenting the new features. Keep in mind that 4.2 is still in beta, meaning everything can still change. This is one reason the documentation is a bit behind.

XK9274, 2023-06-03:
I've been adding new features to these (authentication) and the PR's are still outstanding. 
https://github.com/OnionUI/Onion/pull/867
https://github.com/OnionUI/Onion/pull/877

When everything has been tested/approved I'll write up the new pages for the wiki :)

scottchiefbaker, 2023-06-07:
Excellent, thanks for the response. I'm happy to help test these features, I just need the credentials.

XK9274, 2023-06-07:
The drafts for some these pages are up:

https://github.com/OnionUI/Onion/wiki/HTTP-fileserver
https://github.com/OnionUI/Onion/wiki/SSH-SFTP
https://github.com/OnionUI/Onion/wiki/FTP
https://github.com/OnionUI/Onion/wiki/Telnet

Bare in mind some PR's haven't been reviewed yet so you may not see some of these functions in the current latest build.

XK9274, 2023-08-03:
Closing this as it's being progressed in: https://github.com/OnionUI/Onion/issues/849 and the above requested Wiki pages are now live:

SSH: https://github.com/OnionUI/Onion/wiki/SSH-SFTP
HTTP: https://github.com/OnionUI/Onion/wiki/HTTP-fileserver

If there's any more info you want in the wiki please reopen or raise a new issue. 

Currently the username and password has been changed to a new method and cannot be changed (being discussed)

closed 2023-08-03 by XK9274