OnionUI/Onion #880: L1 and R1 Buttons No Longer Page Up/Down in Game List

Phanturian, 2023-06-03:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion-v4.2.0-beta-dev-3ce1f62

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

The L1 and R1 buttons no longer page up/down or scroll in any list. Pressing L1 or R1 still presents the click, or sound effect associated with scrolling, but nothing happens. L2 and R2 are now assigned to page up/down or scroll, but with no click or sound effect.

### Steps to Reproduce

1. Install Onion 4.2.0 Beta
2. Navigate to a list of games
3. Press L1 or R1

### Provide any additional context or information that may help us investigate your issue

I have the setting to swap L1/R1 with L2/R2 turned off, although this has no effect on the page up/down scrolling in menus.

schmurtzm, 2023-06-23:
Yes Miyoo has changed the page scroll mapping from L1/R1 to L2/R2 and they forgot to move the sound affectation.
This bad sound affectation impacts Onion on this bug : https://github.com/OnionUI/Onion/issues/952
We can hope that it will be fixed on a future MainUI version from Miyoo 🤞

Aemiii91, 2023-06-30:
Fixed in #959

closed 2023-06-30 by Aemiii91