OnionUI/Onion #887: false in-game action on pressing shortcut too fast (delay betwen pressing menu button and locking keys in game)

light-and-ray, 2023-06-06:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

 Onion-v4.2.0-beta-dev-072c73c

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

If I press shortcut "Menu button + SomeKey" too fast, both actions work: shortcut action, and in-game SomeKey action. If I hold Menu button about 100ms and then press SomeKey button, then works only shortcut action, which is correct. I'm sure I press then in right order. If I press they in wrong order the behavior is the same, but in this case it is reasonable

For example in the video 1 first times work both actions: changing aspect ratio and pause in game, and in the last times works only changing aspect ratio

I can do not remove finger from menu button, and the bug is here in first pressing start button. Video 2

This bug is really annoying

### Steps to Reproduce

Press menu button in game and then press some key as fast as possible

### Provide any additional context or information that may help us investigate your issue

https://github.com/OnionUI/Onion/assets/33491867/52f16e38-2781-4e34-8b0a-9d04a2a5b465

https://github.com/OnionUI/Onion/assets/33491867/70cd805f-a678-46c6-9af1-c06cc28b688b

light-and-ray, 2023-06-06:
@Aemiii91 
Can you create an issue for retroarch? It's obviously they made delay for work shortcut in both directions: before and after pressing menu button. But it's needed to have delay only before pressing menu button for case user press buttons in wrong order

I don't know how to formulate it to them and I don't know anything about onion os' code structure

Maybe in retroarch's shortcuts it's not clear which key is menu key. It can be deducted which key is not using in game

Aemiii91, 2023-06-06:
I don't agree there's an issue with this. And I think introducing a delay on *game-related* buttons would be a bad solution.

light-and-ray, 2023-06-06:
@Aemiii91, Hm I could do mistake. There is no any delay after pressing start button and mustn't be. Retroarch knows which button is menu because if press shortcut not so fast It works correctly. They have problem with non instant locking keys after pressing menu button

light-and-ray, 2023-06-06:
The vice versa order of pressing keys is very different case and it might not be connected with the problem. I did mistake

light-and-ray, 2023-06-06:
I think you could formulate this problem in retroarch correctly. I just do not know anything about shortcuts, and about retroarch in general (

closed 2023-07-04 by Aemiii91