OnionUI/Onion #895: Vokume Buttons not working?

Killryde, 2023-06-14:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

I am having isssues not not being able to adjust the volumn with the buttons

### Steps to Reproduce

Hard to say, i think i may be the only one expeirencing this issue since when i click the volumn  button the olumn doesnt go down

### Provide any additional context or information that may help us investigate your issue

N/A

duckyb, 2023-06-16:
Probably a hardware issue, try to boot without an SD card and check if it works in the main menu of MainUI

kj20029, 2023-06-21:
actually i complained about this issue already also and i assumed it was a hardware issue but im guess that if someone else has it, it means there is a bug, @killryde is it like you press the button some time the line shows up on the side but doesnt move and sometimes it just doesnt show up all together

Killryde, 2023-06-21:
It was a software compile  issue(all i did was re install it and it fixed the issue)

closed 2023-06-21 by Killryde