OnionUI/Onion #908: Retro achievements notification sound and icon missing

Fieryfire, 2023-06-15:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2 beta

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

Retroachievements notification sound and icon missing when triggered. Also the verbose mode, to show the available achievements in a level is not showing up.

### Steps to Reproduce

Install ONION OS 4.2 DEV
Load a game with retrologin

### Provide any additional context or information that may help us investigate your issue

Retroarch version could be updated perhaps?

emperordahc, 2023-06-15:
It's not a bug, already posted: https://github.com/OnionUI/Onion/issues/770

closed 2023-07-04 by Aemiii91

Fieryfire, 2023-07-04:
Amazing. Cant wait for the next release to try out the achievements icon and notification. gives thw miyoo mini plus a new life

Fieryfire, 2023-07-04:
Could you also add encore and verbose mode

Aemiii91, 2023-07-04:
I closed this because it was a duplicate. Nothing has been implemented in this regard.

Fieryfire, 2023-07-04:
Oh i see. no worries. we are eagerly waiting. thank you for all the work you put into the OS