OnionUI/Onion #912: No games, apps, etc on latest beta on MM+

mwsduelle, 2023-06-16:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0-beta-dev-b321c9a

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

I installed the latest update manually (not OTA) and after booting up I have the main menu with Recent, Favorite, Game, Retroarch, App, and Settings. Everything besides settings is empty. I was not prompted to install apps when upgrading. I had 15 installed but now there is nothing. My games list is blank (doesn't even say empty). Nothing under Retroarch, Favorites, or Recent. I was not prompted to install any cores or apps during the upgrade like the last few times. Also, select doesn't refresh the game list anymore. I pressed menu and selected "Refresh Roms", though.

### Steps to Reproduce

I was on the previous beta before this and had updated via OTA.

1. Download 4.2.0-beta-dev-b321c9a
2. Extract onto MM+ microSD card
3. Merge/overwrite everything
4. Receive empty categories

### Provide any additional context or information that may help us investigate your issue

[image]
[image]
[image]

schmurtzm, 2023-06-16:
Thanks for your feedback, can you share your .tmp_update\logs\install.log file here ?
For now I can't reproduce but we investigate.

mwsduelle, 2023-06-16:
The .tmp_update folder has disappeared from my microSD after updating...

mwsduelle, 2023-06-16:
I "fixed" it by copying the files from /miyoo/app/.tmp_update ($appdir) to /.tmp_update ($sysdir), inserting the microSD into the MM+, powering on which seems to have started a fresh install. None of my Onion or Retroarch settings were saved, but my saves/states are still there. I looked at the script in the $sysdir and noticed it just executes MainUI in $appdir which ends up copying everything from $appdir to $sysdir. I figured something must have gone wrong in that process considering how it didn't show me the package installer upon upgrade or show any tips during the upgrade process. No idea what happened but I'll be backing up my RA configs before upgrading from now on.

degreeszero, 2023-06-17:
it seems pretty clear to me this latest beta update borked a lot of systems. I ran OTA and now nothing works. I just did a fresh manuak install on a second device and still nothing works.

ronvandegraaf, 2023-06-17:
I have experienced the same issue after OTA.

Phanturian, 2023-06-17:
> I have experienced the same issue after OTA.
> 
> Edit: Maybe it has something to do with .tmp_update folder which has changed to a ._.tmp_update file?

The ._ (aka dot underscore) you have in front of .tmp_update is a file generated by Mac computers. These are problematic with the Miyoo Mini systems, and should be removed before inserting the sd back into the system.  Choosing a file within the Miyoo system that has a ._ before it will usually cause the system to freeze and the battery must be detached to shut down. There is a tool to remove these files within the Tweaks menu, but when you install a new version of Onion you are greeted with the install before reaching the MainUI.

Mac users should open Terminal and type dot_clean followed by a space. Then drag the icon for your sd card drive from the desktop into the Terminal window to add the location of your sd card. Press Return and wait for it to finish. This will remove all the unwanted ._ files from the sd card. Mac users should do this every time they add/edit files on the sd card. It's much faster than using the remove OSX files in the Apps/Tweaks/Tools menu.

andornaut, 2023-06-17:
> I "fixed" it by:
> 
>     1. copying the files from /miyoo/app/.tmp_update ($appdir) to /.tmp_update ($sysdir)
> 
>     2. inserting the microSD into the MM+
> 
>     3. and powering on which seems to have started a fresh install
> 
> 
> None of my Onion or Retroarch settings were saved, but my saves/states are still there. I looked at the script in $sysdir and noticed it just executes MainUI in /miyoo/app which ends up copying everything from $appdir to $sysdir. I figured something must have gone wrong in that process considering how it didn't show me the package installer upon upgrade or show any tips during the upgrade process. No idea what happened but I'll be backing up my RA configs before upgrading from now on.

Thanks, this worked for me too!

Aemiii91, 2023-06-19:
We're now using this issue to collect all bug reports regarding the beta installation fails.

tonyunreal, 2023-06-20:
My problem is a little bit different, I updated through ota and it rebooted straight into MainUI. It behaves exactly as stock ui (recent, device info...all there), none of the programs in the app folder works. 

However, I can launch retroarch fine and the game folder is intact, all the system are there and could be played. Volume change overlay and game switcher are gone, but I could exit a game with menu + select then use the retroarch menu to quit.

mwsduelle's solution didn't work for me cause my /miyoo/app/.tmp_update directory is empty.

Aemiii91, 2023-06-20:
@tonyunreal it might seem different, but the issue is the same: files are not getting unpacked correctly by the installer

Aemiii91, 2023-06-21:
Everyone try the latest update, the issue should be fixed :)

schmurtzm, 2023-06-21:
To apply [this new update](https://github.com/OnionUI/Onion/releases) you can try OTA however as when you'll run it it will be the not fixed version it could fail. So in case of problem [make a classical offline update](https://github.com/OnionUI/Onion/wiki/Installation#upgrading-from-stock-or-onion) for this time, it should be better next time with the updated OTA script 😉

(when doing a manual update, a scandisk on your SD card before copying all the files still a good idea)

If you're a fan of ota and feel courageous you can also update [the ota script](https://raw.githubusercontent.com/OnionUI/Onion/main/static/build/.tmp_update/script/ota_update.sh) manually before running it (located in `.tmp_update/script`).

ronvandegraaf, 2023-06-21:
I tried OTA again. No issues this time. Thanks for the hard work!

schmurtzm, 2023-06-21:
Thanks for then feedback 👍🏻

closed 2023-06-28 by Aemiii91