OnionUI/Onion #914: Add mute function to Global Shortcuts

Taranchul, 2023-06-16:
### Checklist

- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and something is incorrect or missing

### Wiki Section

https://github.com/OnionUI/Onion/wiki/Global-Shortcuts

### Description

The mute function by pressing Vol up & down simultaneously is missing from the shortcut page.

### Correction

Add Vol up+down = mute to the shortcuts.

duckyb, 2023-06-16:
I've added it

[image]

Aemiii91, 2023-06-16:
Thanks @duckyb !

closed 2023-06-16 by Aemiii91

Taranchul, 2023-06-17:
Thanks from me, too!