OnionUI/Onion #917: MM+ goes black

thearti121, 2023-06-16:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0-beta-dev

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

Hi, after installing OS my console won't launch, it only goes black on OS logo, like on video I attached. 
https://streamable.com/txr3a9

### Steps to Reproduce

Before anything I updated firmware to MY354 then I formated brand new sandisk 128gb. Unziped the new 4.2.0 version I checked hiden files in windows and copied all files to card and installed OS. After rebooting console goes black on OS logo. 

### Provide any additional context or information that may help us investigate your issue

https://streamable.com/txr3a9

Aemiii91, 2023-06-16:
Try reinstalling

Aemiii91, 2023-06-16:
Does it boot to stock when you turn it on without SD card?

thearti121, 2023-06-16:
Yes it does

Aemiii91, 2023-06-16:
That's a good sign at least. Did the installation run fine? No sudden exits?

thearti121, 2023-06-16:
None, it installed without errors.