OnionUI/Onion #920: Games not launching

vcuongvu, 2023-06-16:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

I just installed a fresh install of OnionOS.
I've kept all the bios files from the original SD card provided by Miyoo.
I've added some games in the roms folder.
I refreshed the games list and deleted the hidden files but everytime I launch a game it keeps crashing.
When I launch any game from any console it shows :
- Loading
- The screen turns black
- Loading...
And then back to the game list.

### Steps to Reproduce

Fresh install
Adding all bios files
Adding roms
Refresh games list
Delete hidden files
Launching games

### Provide any additional context or information that may help us investigate your issue

No other context

souziago, 2023-06-16:
Same thing here.

XK9274, 2023-06-16:
What happens if you start Retroarch from the Apps page and load content? 
Do other apps load from the Apps page? (Tweaks, etc)
Do you have logging enabled? If so can you send us the logs.

Logging is a hidden menu item.

To enable it:
> Open RetroArch from Apps
> Settings
> User Interface
> Menu Item Visibility
> Settings
> Turn on logging
> Go back to the first Settings page
> Enter logging
> Turn all options on

Reproduce the failure, then go to /.tmp_update/logs and post the RA log files

vcuongvu, 2023-06-16:
When I open the Retroarch app, it crashed just like the games.
It shows Loading, black screen and Loading... and comes back to the apps menu.

Tweaks and Theme changes work normally.

I have three log files :
- batmon => Empty
- install :

:: Check firmware
STATS
Install RA check: 1
Onion total: 3249
RetroArch total: 994
:: Backup system
:: Remove configs
:: Install Onion
   - Extract '/mnt/SDCARD/.tmp_update/onion.pak' (3249 files) into /mnt/SDCARD
:: Install RetroArch
   - Extract '/mnt/SDCARD/RetroArch/retroarch.pak' (994 files) into /mnt/SDCARD
:: Install configs (reset: 1)
[CHECK] checking archive: /mnt/SDCARD/Themes/3DS mini by MarsTaco.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/3DS mini by MarsTaco
[CHECK] checking archive: /mnt/SDCARD/Themes/Analogue by Aemiii91.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/Analogue by Aemiii91
[CHECK] checking archive: /mnt/SDCARD/Themes/CyberOnion (2-pack) by Aemiii91.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/CyberOnion [Atomic] by Aemiii91
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/CyberOnion by Aemiii91
[CHECK] checking archive: /mnt/SDCARD/Themes/GBMini by Kitsuvi.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/GBMini by Kitsuvi
[CHECK] checking archive: /mnt/SDCARD/Themes/Hippo Stock.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/Hippo Stock
[CHECK] checking archive: /mnt/SDCARD/Themes/M by tenlevels.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/M by tenlevels
[CHECK] checking archive: /mnt/SDCARD/Themes/NanoSwitch (2-pack) by Amdy.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/NanoSwitch Dark by Amdy
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/NanoSwitch by Amdy
[CHECK] checking archive: /mnt/SDCARD/Themes/ONION PS (4-pack) by hanessh4.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/ONION PS Dark Blue by hanessh4
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/ONION PS Purple Grey by hanessh4
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/ONION PS Dark Grey by hanessh4
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/ONION PS Blue by hanessh4
[CHECK] checking archive: /mnt/SDCARD/Themes/Onion Boy DX by PixelShift.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/Onion Boy DX by PixelShift
[CHECK] checking archive: /mnt/SDCARD/Themes/Onion Boy by PixelShift + Jeltron.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/Onion Boy by PixelShift + Jeltron
[CHECK] checking archive: /mnt/SDCARD/Themes/Onionboy HD by Jeltron.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/Onionboy HD by Jeltron
[CHECK] checking archive: /mnt/SDCARD/Themes/Oransje by evolve.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/Oransje by evolve
[CHECK] checking archive: /mnt/SDCARD/Themes/PAC-MIYOO by tenlevels.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/PAC-MIYOO by tenlevels
[CHECK] checking archive: /mnt/SDCARD/Themes/RPO by TheDewd.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/RPO by TheDewd
[CHECK] checking archive: /mnt/SDCARD/Themes/Super Onion Entertainment System by TheDewd.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/Super Onion Entertainment System by TheDewd
[CHECK] checking archive: /mnt/SDCARD/Themes/System 9 by injekim.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/System 9 by injekim
[CHECK] checking archive: /mnt/SDCARD/Themes/TechDweeb by TechDweeb.zip
  [EXTRACT] -> /mnt/SDCARD/Themes/.previews/TechDweeb by TechDweeb
MI_SYS_Init cost 2066.000000ms
7779.000000 driver_name (nil)
7780.000000 FB_CreateDevice GFX_Open return 0
init audio 20
FB_EnterGraphicsMode start
iotcl VT_GETSTATE start
iotcl VT_ACTIVATE start 1
iotcl VT_WAITACTIVE start
iotcl KDGKBMODE start
iotcl KDSKBMODE start 3
[... 107 lines cut ...]
FB_EnterGraphicsMode start
iotcl VT_GETSTATE start
iotcl VT_ACTIVATE start 1
iotcl VT_WAITACTIVE start
iotcl KDGKBMODE start
iotcl KDSKBMODE start 3
iotcl KDSETMODE start
iotcl VT_LOCKSWITCH start
FB_EnterGraphicsMode end
[InitKeyShm]: create shm fail : File exists, size 3280
_gfx_alloc_surface 0x350d0, 0xb6419000, idx 0
FB_AllocHWSurface1 0x35060, 0x350d0, 0xb6419000, flags 10001, 640x480, idx 0
SStar_CheckHWBlit
FB_FreeHWSurface 0x35060, 0x350d0, 0xb6419000
_gfx_free_surface 0x350d0 , 0xb6419000
FB_LeaveGraphicsMode A
ioctl KDSETMODE KD_TEXT
ioctl KDSETMODE KD_TEXT 3
ioctl VT_UNLOCKSWITCH
FB_FreeHWSurface 0x341a0, 0xb6ee1af0, (nil)

network :

Thu Jan  1 04:00:03 UTC 1970 Network Checker: Update networking
Thu Jan  1 04:00:03 UTC 1970 Telnet: Killing firmware telnetd process

vcuongvu, 2023-06-16:
I actually ordered 2 Miyoo Mini Plus and both have the same 20230505 firmware.
Both reacted the same way with fresh installs.

vcuongvu, 2023-06-16:
Ok, I'll try in a bit, just a mention :
I tried to install all the apps and all the standalone emulators under Expert and the menu Expert never appears even after reboot.