OnionUI/Onion #921: ScummVM standlone missing GUI scrollbars

vonmillhausen, 2023-06-16:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.2.0-beta-dev-B321C91

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

The Scumm VM standalone app included with recent dev builds lacks GUI scrollbars - this makes it impossible to scroll down to settings hidden below the visible portion of the GUI window. As an example, it makes it impossible to set keybinds for games within the GUI, as only the first few keybinds are accessible. The current Retroarch core version of ScummVM _does_ include scrollbars

### Steps to Reproduce

1. Launch the standalone ScummVM app, via any game
2. Press Start to bring up the ScummVM menu, and choose "Return to Launcher"
3. With a game selected in the list, choose "Game Options..."
4. Switch to the "Keymaps" tab
5. Note that only the first few mappings are visible, with the rest inaccessible below the GUI "fold" due to the lack of scrollbar

### Provide any additional context or information that may help us investigate your issue

Here's what the Retroarch core version looks like in the Keymaps tab for the game "Lands of Lore: The Throne of Chaos" - note the scrollbar on the right which allows you to scroll down and see all of the available keymaps for the game:

[image]

Here's what the standalone version looks like in the same menu for the same game - note the scrollbar is now missing, so it's no longer possible to access all the keymap settings (they default keymaps in the standalone version for this game are such that the game is unplayable):

[image]

XK9274, 2023-06-16:
Hey,

This is a known issue and is inherited from the scummvm source as documented in the PR for ScummVM: https://github.com/OnionUI/Onion/pull/904

schmurtzm, 2023-08-01:
Should be fixed in the fresh 2.7.1 released officially. ;)

XK9274, 2023-08-02:
https://github.com/OnionUI/Onion/pull/1092 has merged, will be in next release or can be manually installed by downloading the binary.

closed 2023-08-02 by XK9274