OnionUI/Onion #925: OTA Update - Github Not Reachable

redmarkred, 2023-06-17:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion-v4.2.0-beta-dev-b321c9a

### Miyoo Firmware Version

202303021817

### Provide a clear and concise description of the issue

OTA update give an error - "https://github.com not reachable. Check your wifi connection"

Have confirmed that the wifi connection is fine by connecting via SSH/FTP etc and tried on various Wifi networks too.

### Steps to Reproduce

Boot Onion OS, check wifi is connected, launch OTA Update app.

### Provide any additional context or information that may help us investigate your issue

Running on a Miyoo Mini Plus bought a month or so ago

XK9274, 2023-06-17:
Your firmware may not have DNS support, there were some earlier ones that didn't. 

You can confirm this by doing: `nslookup github.com` in terminal, if you get an error that's why OTA is failing.

Upgrade your firmware to: https://discord.com/channels/529983248114122762/1063196114448289792/1091250540148703262

redmarkred, 2023-06-17:
Thank you! I updated to firmware version 202305052130 and this has solved my issue. The discord link you posted didn't work for me but I found the firmware on the rather sketchy but apparently legit miyoo website and updated to that - www.lomiyoo.com/en/

Thanks so much for your help and hard work on this!

XK9274, 2023-06-17:
Awesome thanks for reporting back! No worries & my bad for the link, seems to work for me. Maybe you're not in the RGD discord server!

closed 2023-06-17 by XK9274

redmarkred, 2023-06-17:
No worries, totally my fault with the link... I wasn't in the RGD server. Discord is a bit confusing to me but I joined the server and it's showed up! Thanks again for your help!