OnionUI/Onion #929: InstallUI optimizations

schmurtzm, 2023-06-18:
InstallUI consumes a lot of CPU and memory : 

[image]

This can triggers unexpect crashs / reboot during install process.

Removing `miyoo/app/.tmp_update/res` folder helps : installUI is using 60MB instead of 3GB of virtual memory and no more strange reboot during install extracts. 

It looks like there is a memory leak because the memory consumption growth during all the install (start at around 20MB and stop at the end of the install at 70MB) : I see that each time it slides an image the memory consumption increase of 8MB (even without any media present) if you press Left or Right many time the memory increase very fast !

tGecko, 2023-06-20:
Was improved a lot by #931 by not calling `imageCache_getItem` every run of the `while (!quit)` loop.

The underlying issue still exists in [imageCache](https://github.com/OnionUI/Onion/blob/main/src/common/utils/imageCache.c)
but finding it is out of my league.

Additionaly there is still a bug in installUI.c:

In line 202 it checks `if (acc_ticks >= time_step)` but `acc_time` never gets reset to zero so it keeps on rising, so this check always equals to true after acc_ticks >= time_step for the first time.

I did attempt to fix this by replacing `acc_ticks -= time_step;` with `acc_ticks = 0;` in line 239, and it seems to behave properly then.
But I could not measure a performance increase by doing so, presumably because the loop sleeps for 15msec anyway (`msleep(15);` in line 242) so I did not include it in the PR.
Install time of the system was the exact same before and after the change.

Aemiii91, 2023-06-20:
@tGecko `acc_ticks -= time_step;` will eventually hit zero (should hit zero every frame - but due to delays it can vary)

tGecko, 2023-06-20:
@Aemiii91 this is what I get with `sprintf(message_str,"acc_ticks: %d",acc_ticks);` inserted in line 226:
https://streamable.com/3zcxzp

Aemiii91, 2023-06-20:
@tGecko wow.. didn't know it was delayed that much, doesn't even go down 😵‍💫

Aemiii91, 2023-07-04:
Closing as done. Since #956 installUI's memory usage is stable.

Related PRs:
- #931 
- #956 / 8c779147c815322992f69b96e4a7c5d7eb3dc9aa

closed 2023-07-04 by Aemiii91