OnionUI/Onion #933: Broken upgrade during latest OTA installation

Amiga500, 2023-06-19:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

OTA update of this night from latest one

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

I have made as usual an OTA upgrade, but this time after downloading the release and the reboot, after about 1 minutes the miyoo reboots without finish the installation process and onion was a mix between it and stock os. It was no possible to relaunch OTA upgrade, going to theme and putting default theme, it put the stock theme.. and so on..

### Steps to Reproduce

go to OTA upgrade, launch it and wait

### Provide any additional context or information that may help us investigate your issue

.

XK9274, 2023-06-19:
Do you know what version of onion you updated from?
How far did it get into the install?
On your SD are you missing .tmp_update?

I will try reproducing tonight w/ more logs.

Checklist

geoffclifton, 2023-06-19:
I experienced the same issue. Please let me know if there are specific logs I can pull off my device to help with the investigation.

Amiga500, 2023-06-19:
Do you know what version of onion you updated from?
It was the latest beta previously of this one, i don't remember the tag

How far did it get into the install?
Maybe 2/9 or 3/9, about 1 minute from the reboot

On your SD are you missing .tmp_update?
Yes, and into update.log there is wrote: start updating again
[image]

ronvandegraaf, 2023-06-19:
I experienced the same issue (same version, missing the .tmp_update after the update) with update.log stating "start updating again".

I manually downloaded the latest vision on my laptop and copied to my sd-card to force a new install. Settings where lost.

Aemiii91, 2023-06-19:
Duplicate of #912

closed 2023-06-19 by Aemiii91