OnionUI/Onion #934: All options in "App" fail to load after brief "Loading" screen

thediamondgeezer, 2023-06-19:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

96c87c3

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

All options in "App" fail to load after brief "Loading" screen: Tweaks, Search, RetroArch, Package Manager, Onion OTA Update fail to load after brief "Loading" screen then return to App selection menu.

### Steps to Reproduce

Navigate to "App" from the main menu, press "A" to select an entry.

### Provide any additional context or information that may help us investigate your issue

Happens since upgrade new beta

dreams897, 2023-06-19:
Had this problem as well, can't launch Retroarch with latest build

Aemiii91, 2023-06-19:
Duplicate of #912

closed 2023-06-19 by Aemiii91