OnionUI/Onion #942: Is it normal for onionos updates to wipeout user settings?

Plloi, 2023-06-21:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0-beta 41caa40

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

After running the ota update app...
My retroarch color scheme reset to default and the one i was using just seems to be gone
The HTTP server password also reset to default
My SNES default core changed (not wholly unexpexcted since i had tampered with the launch.sh )

### Steps to Reproduce

Apparrently just customize and update from the older 4.2.0 beta

### Provide any additional context or information that may help us investigate your issue

Not sure if this is OTA specific or not, it's how i updated.

Aemiii91, 2023-06-21:
RA theme and `Emu/RApp/App` folders will all get reset on update, that's normal. We are you into fixing the file browser credentials being deleted 

We might be at a stage where we can remove the forced RA theme (it was mostly for users upgrading from <4.0)

XK9274, 2023-06-21:
Hey! 

The database unfortunately does store the credentials but i've stored it in a none-persistent location, I'll raise a new PR to move this into a persistent location that won't be reset by upgrades. 

Will be fixed in next release (but you'll likely lose your DB again moving to this release so i'll also provide instructions to back it up and restore after the upgrade!)

Plloi, 2023-06-21:
> Will be fixed in next release (but you'll likely lose your DB again moving to this release so i'll also provide instructions to back it up and restore after the upgrade!)

Maybe have a few cycles where the update script checks the old location for a db an moves it to the new one?

XK9274, 2023-06-22:
https://github.com/OnionUI/Onion/pull/948

PR to address this issue.

closed 2023-07-04 by Aemiii91