OnionUI/Onion #943: WIFI will not connect to SSID with "#" in the name

Milky28, 2023-06-21:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.2.0-beta-dev-96c87c3

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

Received the "Fail to connect Check your password or reboot router" error when attempting to connect to a WIFI access point that contains a "#" character in the SSID.

Note - to be clear this is for special characters in the _SSID_, not the password as has been previously reported.

### Steps to Reproduce

Turn on WIFI --> select SSID with a # in the name --> enter password --> "Fail to connect Check your password or reboot router" error received.

### Provide any additional context or information that may help us investigate your issue

After creating a new SSID without the "#", it connects as expected.

XK9274, 2023-06-21:
Can you check the wiki page [here](https://github.com/OnionUI/Onion/wiki/Frequently-Asked-Questions-%28FAQ%29#my-wifi-wont-connect-as-it-contains-special-characters-what-do-i-do)

I've added a new entry, let me know if you need further guidance so I can assist & update the wiki so others can benefit.

Milky28, 2023-06-21:
> Can you check the wiki page [here](https://github.com/OnionUI/Onion/wiki/Frequently-Asked-Questions-%28FAQ%29#my-wifi-wont-connect-as-it-contains-special-characters-what-do-i-do)
>
> I've added a new [...]

That entry was helpful, thank you.  The third party app was a good workaround for me.

XK9274, 2023-06-21:
Great! thanks for the feedback & glad the app worked for you.