OnionUI/Onion #950: No netplay option from Recent menu

georgesak, 2023-06-22:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0-beta

### Miyoo Firmware Version

202306092122

### Provide a clear and concise description of the issue

From the Recent menu, going in  GLO, no option to start netplay.

### Steps to Reproduce

Enable recent menu
Select game from the menu
Press Y (go in GLO)
No netplay option 

### Provide any additional context or information that may help us investigate your issue

N/A

XK9274, 2023-06-22:
Hey,

Do you mean the new feature (start netplay over hotspot) or are you referring to something else?

georgesak, 2023-06-22:
Yes, that's what I meant. The wifi direct option (which is awesome btw :).

XK9274, 2023-06-23:
Hey,

Thanks for the feedback! 

Check the notes/comments on  https://github.com/OnionUI/Onion/pull/891 for a detailed explanation and a temporary solution.

It's something I'm aware of I just need to have a chat with the other guys how best to do it without duplicating too much.

georgesak, 2023-06-23:
Awesome, thanks!

Aemiii91, 2023-07-04:
Done in:
- #969

closed 2023-07-04 by Aemiii91