OnionUI/Onion #957: Issue Title

Elgeweily, 2023-06-26:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0 Beta

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

Save states not working in Prince of Persia 1 & 2 for both DOSBox Pure 0.21 & 0.9.7, running on RetroArch 1.15.0. When I try to Save, Load, or Resume, the whole game restarts. Except for specific parts of the game where Save and Load work correctly, for example: during the game intro and at the end of level 1, other than that it doesn't work, and the game restarts.

### Steps to Reproduce

On Miyoo Mini+ with firmware: 202305052130, install Onion OS 4.2.0 Beta, Enable DOSBox Pure Core (default is Version 0.21), Run Prince of Persia 2 DOS Version Rom, and try to save anywhere except in the intro or the end of level 1, and the game will restart.

### Provide any additional context or information that may help us investigate your issue

I tried the same game (POP2) on the stock OS, running on DOSBox Pure 0.9.7, it is horribly slow but it is saving and loading correctly.

Also, sometimes when I close POP2 on Onion OS, it exists with Error Code: 139.

XK9274, 2023-08-02:
Honestly one of the most infuriating games i've ever played. Where do you get up to when saves stop working? I made it to the beach and the icons in the manual and gave up but save states worked all the way up until after the quicksand bit. IDK maybe my binds are broken for DOSbox pure..

Elgeweily, 2023-08-03:
When the save stops working, it goes back to the beginning of the game, to
the intro....actually I solved the issue by using the assigned hotkeys for
saving and loading, it works fine, when you use the retroarch menu save and
load, it doesn't work.

XK9274, 2023-08-03:
Ah ok, then i can't reproduce this. I'm at the beach stage and loading/saving with either RA menu or the hotkeys menu/bumper both work for me.