OnionUI/Onion #986: Setting Brightness in Settings cannot go below 01/10, but can using hotkey

theUnBurn, 2023-07-05:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.2.0-beta-dev-41caa40

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

Brightness control in Settings does allow you to go to 00/10, but the hotkey does (i.e using MENU + Volume Down).

### Steps to Reproduce

1. Go To Settings -> Brightness
2. Press left on Brightness until you get as low as you can (01/10)
3. Use the hotkey to change Brightness down, showing now that you are able to set this value to 00/10

Expected: Brightness should be able to go down to 00/10 using the Brightness Settings in Settings.
Actual: Brightness cannot go below 01/10

### Provide any additional context or information that may help us investigate your issue

https://github.com/OnionUI/Onion/assets/63260253/655f463e-f242-4051-aff7-605dc708ab84

theUnBurn, 2023-07-05:
Super duper small issue, but figure I'd document it in case anyone gets around to it!

Aemiii91, 2023-07-05:
Duplicate of #866

closed 2023-07-05 by Aemiii91