OnionUI/Onion #1016: Vsync issue

Refaze, 2023-07-18:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion beta 4.2 (latest)

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

May not be a bug in Onion but I get alot of screen jittering.  Enabling Vsync fixes this issue but causes audio quality to drop significantly.

### Steps to Reproduce

Enabling Vsync.

### Provide any additional context or information that may help us investigate your issue

I have 3 minis all experiencing the same issue.  Vsync is disabled by default.  Screen jitter is mostly seen on more taxing cores,  GBA and PSX.

Aemiii91, 2023-07-18:
What can we do about it?

Aemiii91, 2023-07-18:
You say you have 3 Minis, but that you're on fw 0505, which is Mini Plus fw.. so which is it?

Also, would be nice with some video examples.

Refaze, 2023-07-18:
It is very hard to catch on camera.  To the eye the scrolling just isn't smooth.  I can live with it I just wanted to make sure my units were not bad.  It seems to be more profound on GBA titles.  Vsync is an instant fix but it drops the sound quality.

esotericode, 2023-07-18:
I can confirm I see the same thing. I don't think there's anything wrong with your device. 

I can easily see the issue by running a game like Pokemon LeafGreen on GBA, and there's a noticeable stutter when walking and the screen scrolls with v-sync disabled (which is indeed the default it seems) and it's perfectly smooth with  v-sync flipped on. 

Some people are more sensitive to these kinds of issues than others, me personally I tend to notice them and it drives me crazy. However, I think most people don't really notice it or aren't very sensitive to it. 

The reason audio might sound worse with v-sync enabled is that some emulators like GBA or PS1 are pretty heavy and if you enable v-sync the frame rate dips are more apparent and tend to affect the audio more obviously. I could be quite wrong though, and I'm sorry if I am completely off base. I'm not an expert on this stuff.

Refaze, 2023-07-18:
Your not off base at all this is exactly it as PSX and GBA are more taxing.  I can live with the scrolling was just worried it was my device.  It's great to know this is most likely normal.

Refaze, 2023-07-18:
Thank you for the comparison so its not an onion issue at all.  I can close the post.  Thanks.

closed 2023-07-18 by Refaze

Refaze, 2023-07-18:
Great news,  I fixed it without dropping performance on the PSX and GBA cores.  I turned on Vsync and used Interval 2 for Vysync.  Been playing for a few hours no hiccups.  I also changed the audio resampler to normal as onion defaults it to low.  Give it a try I would love to hear your feedback.

esotericode, 2023-07-19:
Changing the v-sync interval doesn't seem to actually do anything at all. In fact, I suspect the feature is simply unsupported on this platform and just doesn't have any effect. 

You can easily test this for example by running Retroarch on Windows. If you set v-sync interval to 2 it will basically half the refresh rate and make your game run at half speed. It's quite noticeable! 

On the Miyoo Mini the settings doesn't do anything. In fact you can change it to any other number and it doesn't have any effect. 

Perhaps some other setting you changed got it working better, not sure but I'm glad it's working better. I think it's also going to be pretty dependent on the game as well if you're getting good enough performance out of it. Maybe the sound setting did was helpful too. 

As a side note, I do think v-sync should probably be on by default IMHO.

Herbie85, 2025-08-16:
If anyone should read this in the future, I was able to completely solve the problem with bad audio by setting the vertical refresh rate to 59,600 Hz with VSYNC enabled. 

Retroarch Main Menu => Settings => Video => Output => Vertical Refresh Rate => set from 60,000 Hz to 59,600 Hz