OnionUI/Onion #1023: Can't run "elektronika bk" rom

mikhailter, 2023-07-19:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1.4

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Can't run any "elektronika bk" rom - failed to load content, BIOS files placed.

### Steps to Reproduce

1. Install Elektronika BK via packet managert with expert tab.
2. Add BIOS files to /BIOS/bk folder https://docs.libretro.com/library/bk/
3. Add any rom to /Roms/EBK/

Try to load a rom via Expert > Elektroinikabk and it will not run.

### Provide any additional context or information that may help us investigate your issue

Works fine with latest RetroArch on other devices.

XK9274, 2023-07-22:
This is likely fixed by https://github.com/OnionUI/Onion/commit/b150ae480899d182b5776346b5bd4950ce2861f7

A user was having an identical issue with Mame2000 and using the new runtime.sh in the link solved it for them.

XK9274, 2023-08-02:
Just tested this on `v4.2.0-RC-snapshot-a5064207` and it's working through expert provided you have all the correct BIOS files.

mikhailter, 2023-08-03:
Confirm that it's working on v4.2.0-RC-snapshot-a5064207

closed 2023-08-03 by mikhailter