OnionUI/Onion #1024: Vertical 3x4 Filter Forces Keep Aspect Ratio - ON

Phanturian, 2023-07-19:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion-v4.2.0-RC-snapshot-4d182631

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

Using the vertical3x4.filt now forces the setting for Keep Aspect Ratio to the ON position. It is persistent, and will not be retained if saved in an override. Keep Aspect Ratio used to be forced to OFF with the vertical3x4.filt, which is how it should be.

### Steps to Reproduce

Apply the vertical3x4.filt to a game, go to settings>video>keep aspect ratio and turn it OFF. Save the game override, quit Retroarch, and relaunch your game. Keep aspect ratio will be turned ON automatically.

### Provide any additional context or information that may help us investigate your issue

I saved configs for numerous vertical arcade games, and now they all launch with black bars on the top and bottom. The purpose of the vertical3x4.filt is to stretch the image to the 3x4 ratio, not shrink the original aspect ratio so it fits within the horizontal space of the 3x4 ratio. I can't think of one reason to use the vertical3x4.filt and keep aspect ratio on at the same time.

[image]
[image]

Aemiii91, 2023-07-19:
Just tested the filter on PicoDrive, it doesn't affect the "Keep Aspect Ratio" setting

Phanturian, 2023-07-19:
I found new steps to reproduce. Launch a game and press Menu + Start to change aspect ratio to Integer scaling: OFF (Original) first. Now if you try to add the vertical3x4.filt and manually change Keep Aspect Ratio to OFF and save override, it won't save the "OFF" setting. You must press Menu + Start to return to Integer scaling off (3x4). I believe this behavior was introduced with the autosave override with aspect toggle that was just implemented. The Menu + Start setting seems to take precedent over any manual save override attempts, which was new and unexpected to me.

An option to toggle the autosave override would be nice. I prefer to save manually, but understand the convenience for others. Or allow a manual override save to overwrite the override save created by pressing Menu + Start.

Phanturian, 2023-07-19:
I just now realized that pressing Menu + Start saves the setting in the core override. I use four different cores for my Arcade build which is why I was getting mixed results from game to game. For anyone trying to have custom settings on a per game basis, I recommend using the settings>video>keep aspect ratio and do a manual game override. Otherwise, pressing Menu + Start changes the setting for every game using that core.

Aemiii91, 2023-07-19:
Yes, and if a game override with scaling options or a directory override already exist, <kbd>MENU</kbd>+<kbd>START</kbd> will save to that instead.

closed 2023-07-19 by Aemiii91