OnionUI/Onion #1027: Syncing Time Bug and Suggestion

MichaelRPowell, 2023-07-19:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.2.0-RC-snapshot-4d182631

### Miyoo Firmware Version

202306092122

### Provide a clear and concise description of the issue

Two-part issue.

1. Can there be a way to skip the time sync, or at least show a countdown until timeout? I assume the timeout is set to around 30 seconds from my own experience. If someone connects to a network that is preventing the time sync (no external access), then they are effectively trapped without knowing when they can proceed to the main menu (or how to proceed). Either showing a countdown until it times out or providing a way to skip after 5 seconds would be nice.

2. Time sync seemed to not work for me, I am not entirely sure why. I haven't dug into the code to see how it is being set (commands and checks), but while my Miyoo Mini Plus connected to the local network quickly, it never synced the time. I confirmed the time failed to sync by checking under Tweaks->Startup. I was able to quickly set the time correctly by running my own script which used this command:

`ntpdate pool.ntp.org`

To me that makes it seem like the network was working, and the time could be sync, but the startup time sync process failed for some reason.

### Steps to Reproduce

Try booting the device on a network without external/internet access. You'll find yourself trapped (momentarily) with no clear way to know if it will continue after a timeout or how long that is. Again, a way to skip or a countdown would be great for casual users.

### Provide any additional context or information that may help us investigate your issue

I am not sure why the built-in time sync didn't work, but the ntpdate command did. I will try to look at the commands myself that are used during the start and try to log why that isn't/wasn't working for me.

XK9274, 2023-07-19:
Some of this is likely being addressed in https://github.com/OnionUI/Onion/pull/1013

MichaelRPowell, 2023-07-19:
> Some of this is likely being addressed in #1013

Thank you, that looks exactly what I was talking about. I did a search for open issues since the time didn't sync correctly on boot, probably why I missed that. Those additional changes, feedback to the user, and lower timeouts look great!

closed 2023-07-21 by Aemiii91