OnionUI/Onion #1030: No Polish language in Retroarch

kashkash18, 2023-07-20:
Please add Polish language to Retroarch.

tGecko, 2023-07-20:
It's there - In RetroArch, go to Settings -> User -> Language and select Polish

tGecko, 2023-07-20:
In fact, all [these languages](https://github.com/OnionUI/RetroArch/blob/ef98978914273fe0d9cafae300293e9293c3cd0d/menu/drivers/rgui.c#L1545) don't work:
```c
      case RETRO_LANGUAGE_ESPERANTO:
      case RETRO_LANGUAGE_POLISH:
      case RETRO_LANGUAGE_TURKISH:
      case RETRO_LANGUAGE_SLOVAK:
      case RETRO_LANGUAGE_CZECH:
      case RETRO_LANGUAGE_HUNGARIAN:
```

tGecko, 2023-07-20:
Can you try this please:
Extract the files in this zip and move the two included files to your SD into the folder
`\RetroArch\.retroarch\assets\rgui\font\`
[font.zip](https://github.com/OnionUI/Onion/files/12105237/font.zip)

commit 4b0f959f (2023-07-20): FIX: Added alternative fonts for other UI languages to RetroArch (#1032)

Add missing font files needed for several languages.

Fixes #1030
PR #1032 (2023-07-20): FIX: Added alternative fonts for other UI languages to RetroArch
Add missing font files needed for several languages.

Fixes #1030
closed 2023-07-20 by Aemiii91