OnionUI/Onion #1036: Brightness OSD and Favourite issues.

fengfrw, 2023-07-21:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion-v4.2.0-RC-snapshot-ba4ba37d / Onion-v4.2.0-RC-snapshot-4d182631

### Miyoo Firmware Version

miyoo354-202306092122

### Provide a clear and concise description of the issue

1. The shortcut key for brightness adjustment is invalid, and the OSD bar is not displayed. The OSD bar for volume adjustment is displayed normally.
2. There is no content in the favorites. Adding favorites is normal, but the view Favorites bar is blank. Favourite.json file has the contents.

### Steps to Reproduce

1. Shortcut keys adjust the brightness.
2. Add Favorites and view the Favorites bar.

### Provide any additional context or information that may help us investigate your issue

Return Onion-v4.2.0-beta-dev-41caa40 normal.

Aemiii91, 2023-07-21:
The brightness shortcut is <kbd>MENU</kbd>+<kbd>VOL</kbd> for Mini Plus.

Can you please provide a copy of the favourite.json file.

fengfrw, 2023-07-21:
1.Thank you for your answer. I found that using MENU+VOL conflicts with my long press menu settings (I will reconfigure the menu shortcut to avoid it).
2.I found that the file also became an empty file. Like I don't have a collection. I will test and observe again and wait for my feedback.

fengfrw, 2023-07-21:
> The brightness shortcut is MENU+VOL for Mini Plus.
> 
> Mini Plus的亮度快捷键为MENU+VOL。
> 
> Can you please provide a copy of the favourite.json file.
> 
> 请提供favourite.json文件的副本。

1.Thank you for your answer. I found that using MENU+VOL conflicts with my long press menu settings (I will reconfigure the menu shortcut to avoid it).
2.I found that the file also became an empty file. Like I don't have a collection. I will test and observe again and wait for my feedback.

Aemiii91, 2023-07-21:
It shouldn't conflict. You just have to press <kbd>VOL</kbd> in less than 300 milliseconds.

fengfrw, 2023-07-21:
1.Thank you for your answer. 
2.I found that the file also became an empty file. Like I don't have a collection. I will test and observe again and wait for my feedback.

bmrysz, 2023-07-21:
I found, that shortcut Select+L2/R2 doesn't work either.
firmware/os version: 202306092122/4d182631

Aemiii91, 2023-07-21:
@bmrysz it's not an error.

fengfrw, 2023-07-21:
The collection problem should be the system problem on my mini+. It is normal to test the mini, and it is normal to replace the TF card using the mini with the mini+.

fengfrw, 2023-07-21:
The collection problem should be the system problem on my mini+. It is normal to test the mini, and it is normal to replace the TF card using the mini with the mini+.
Thank  Aemiii91,Thank Onion Team！

closed 2023-07-21 by fengfrw