OnionUI/Onion #1046: change.wav volume increases

TooGeekCreations, 2023-07-22:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

V4.2.0-RC-Snapshot-BS4BA37D

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

charge.wav works fine on the main menu, if I go to the Applications menu iside Tweaks the volume for charge.wav increases maybe doubles. In any submenus in Tweaks no matter where you are the volume is still louder than normal. If you quit Tweaks menu the change.wav volume goes back to normal.

### Steps to Reproduce

See above

### Provide any additional context or information that may help us investigate your issue

Menu sound set to 10 from 20 for me in settings.

XK9274, 2023-08-03:
Have you tried updating to the newest release? I'm on `v4.2.0-RC-snapshot-a5064207` and can't reproduce - if possible can you update and share a video if the issue still exists? Cheers

Aemiii91, 2023-08-03:
I am able to hear a difference (when bgm vol in settings is set to 10)

Aemiii91, 2023-08-03:
(In Tweaks, and all our other apps, we are not using the same volume formula as MainUI)

TooGeekCreations, 2023-08-03:
> (In Tweaks, and all our other apps, we are not using the same volume formula as MainUI)

Oh, interesting...

> Have you tried updating to the newest release? I'm on `v4.2.0-RC-snapshot-a5064207` and can't reproduce - if possible can you update and share a video if the issue still exists? Cheers

I did update just now, oh boy now the applications are in different order what was before. (looks like random to me instead of alpabetic order) But the volume "problem" did not changed. Try it out to set the menu volume 02 and go in to the tweaks menu... the change.wav volume still louder there.

PR #1495 (2024-05-09): Fix discrepancy between MainUI and Onion volume
Use the same algorithm to calculate the volume as MainUI

`(x << 7) / 20` is equivalent to `(x * 128) / 20` which is just plain linear scaling

Comparison:
[image]

Fixes #1046
commit 4b93b700 (2024-05-09): Fix discrepancy between MainUI and Onion volume (#1495)

Use the same algorithm to calculate the volume as MainUI

`(x << 7) / 20` is equivalent to `(x * 128) / 20` which is just plain
linear scaling

Comparison:

[image]

Fixes #1046
closed 2024-05-09 by Aemiii91