OnionUI/Onion #1096: MM+ frozen at Miyoo splash boot screen

rafaelvedovatto, 2023-08-03:
Onions OS Version
4.2.0-RC

Miyoo Firmware Version
202305052130

Provide a clear and concise description of the issue
Hi, once OnionOS installation is complete, the system reboots but get stuck at Miyoo splash boot screen. I'm using a SanDisk 64GB microsd format in FAT32. Here is a video about it:

Steps to Reproduce
Unziped the new 4.2.0-RC version I checked hiden files in windows and copied all files to card and installed OS. After rebooting console goes black on OS logo.

Provide any additional context or information that may help us investigate your issue
https://streamable.com/txr3a9

Please create a new post describing the issue in detail. And if you can, supply a video showing the problem.
_Originally posted by @Aemiii91 in https://github.com/OnionUI/Onion/discussions/919#discussioncomment-6605875_

TooGeekCreations, 2023-08-03:
It was the same with me... does not do anything if i press any button incuding the power. Only option was for me to take the battery and the memo card out then battery back start the original system than power off memo card in, power on and it was working normally started the update.

rafaelvedovatto, 2023-08-03:
I can power off with the power buttom and if I switch back to the stock microsd and stock firmware the system boots just fine. The ROMS folder is empty because I never got to that point in the installation where I can choose which emulator I want to enable/use.
You can take a pick of what's going on in this video uploaded to youtube:
https://www.youtube.com/watch?v=6bTw1W6uw9U

rafaelvedovatto, 2023-08-07:
@Aemiii91 have you had the chance to look into this? Could it be a recurrence of the old problem discussed [here](https://github.com/OnionUI/Onion/discussions/919#discussioncomment-6605875)

Aemiii91, 2023-08-07:
My best guess is you have a bad SD card. The installation is only supposed to take a couple of minutes (not 15)

rafaelvedovatto, 2023-08-09:
:(

I'll get another microsd later today and attempt to install again. Thanks for checking.

yurineves92, 2023-08-28:
I had the same problem, did changing the card solve it?

yurineves92, 2023-08-31:
I changed the card it worked.

rafaelvedovatto, 2023-09-12:
> I had the same problem, did changing the card solve it?
> 
> @rafaelvedovatto

Yep. The root cause was a bad microsd.

closed 2023-09-12 by rafaelvedovatto

JhonatanMontanheiro, 2024-04-26:
I'm having the exact same issue. It wasn't finishing the install properly at first, using the 4.2 stable version. Then I tried the beta and it went all the way to 100% but go stuck on reboot. I'll just get a new card. Thanks for posting about it.

Pxdrxoo, 2024-11-26:
Hi, i´m having the same problem. I tried everything but still frozen on Miyoo screen...

rafaelvedovatto, 2024-11-26:
Most likely you bought a fake or a bad memory card. Get a new one and you should be good.