OnionUI/Onion #1099: games not show up on MMP

Hai1979, 2023-08-03:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

onion v4.2.0-rc

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

the game I downloaded does not show up on the my device after I refreshed it

### Steps to Reproduce

1. Put the roms in the card 
2. look it up on roms folder 
3. did not see it
4. refresh the page where it goes on MMP
5. did not see it

### Provide any additional context or information that may help us investigate your issue

try to reflash and put the same rom on the card it did not show up on the MMP

Aemiii91, 2023-08-03:
Need more details:
- name of rom folder
- name of system
- file extension

Hai1979, 2023-08-03:
GBA folder , GBA like final fantasy 6 advance #GBA.GBA

Aemiii91, 2023-08-03:
Can you send a screenshot of the file.

Hai1979, 2023-08-03:
[image]

Hai1979, 2023-08-03:
here is the screen shot on my laptop

Aemiii91, 2023-08-03:
OK, try renaming it "`Final Fantasy IV Advance.gba`"

Hai1979, 2023-08-03:
It doesn’t work and I missing couple of games on my MMP

Aemiii91, 2023-08-03:
How did you "refresh the page" ?

Hai1979, 2023-08-03:
Games, Apps and then refresh the page

Aemiii91, 2023-08-03:
What do you mean by Apps?

Hai1979, 2023-08-03:
at the main menu screen on my MMP

Aemiii91, 2023-08-03:
OK, then you press <kbd>SELECT</kbd> -> **Refresh all roms** ?

Hai1979, 2023-08-03:
> OK, then you press SELECT -> *Refresh all roms* ?

I press SELECT and press Refresh all Roms , and did not work at all

Aemiii91, 2023-08-03:
Do you have an `.xml` file in your rom folder?

Hai1979, 2023-08-03:
How do I do that?

Hai1979, 2023-08-03:
I do have that on my card

Aemiii91, 2023-08-03:
Why? What's it called? Try removing it.