OnionUI/Onion #1104: Update mGBA core.

twoco, 2023-08-05:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.2.0-RC-snapshot-a5064207

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

The current mGBA core 0.10-dev `ec5ecb2` has an issue of flickering sprites. The commit is from Oct 25, 2022. I would test the latest version 0.10.2 `2fb5545` from Apr 24, 2023. But I have no idea how to update it. Or who compiles this for the Miyoo Mini? Is this the right place here?

### Steps to Reproduce

Run the game "Mario And Luigi Superstar Saga" in mGBA and see flickering sprites e.g. in the intro scene. 

### Provide any additional context or information that may help us investigate your issue

To fix the issue, we have to test with the latest issue. Maybe this will resolve the problem or we file a bug report.

XK9274, 2023-08-05:
Please test this: https://github.com/OnionUI/Onion/pull/1105

twoco, 2023-08-05:
Yes, unfortunately it‘s not fixes. The sprites are still flickering. And moreover the game runs now in stop motion most of time. Unplayable. Anyway, thanks for compiling and testing too. …

TwisX, 2024-05-21:
Installed Onion on my MM+ the other day and am experiencing the same issue with sprites flickering in Mario & Luigi Superstar Saga with the mGBA core. gpSP does not have this issue and seems to be running fine. I suppose it is a non-issue since gpSP works, but it would be great if this wasn't an issue with the "out of the box" Onion install. I'd be happy to help test a newer mGBA core if needed.

Gusfit1, 2024-12-15:
@TwisX

It's an issue with Game Boy Player/Rumble that's enabled by default in core settings. Disabling that fixes it.

Best way I've found to apply the change:

* Load game
* Quick Menu > Core Options > Input & Auxiliary Devices > Game Boy Player Rumble (toggle off)
* Close Content (in Quick Menu)
* Quit Retroarch
* Navigate back to the game and open the GLO menu
* Reset game

You'll see the changes have taken effect. I'm not sure where this saves to, I thought at first that it would be saved to one of the override config files (gba.cfg, mgba.cfg) but no new entries were added when saving those.

This setting should really be set to off by default imo. More people are likely to complain about sprite flickering over lack of rumble in a GBA game, and if they choose to enable rumble the warning is right there.