OnionUI/Onion #1116: Keyboard iper case in scraper

Pisuke74, 2023-08-16:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0beta

### Miyoo Firmware Version

20230505

### Provide a clear and concise description of the issue

When i insert my scraper fr account the keyboard doesen t accept upper case using shift or caps lock 

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue
Scrape a game

### Provide any additional context or information that may help us investigate your issue

None

poc-sm, 2023-08-26:
Hello 
Same issue here + it's not really useful sometimes awkward caracteres displayed 
More efficient to edit config file manually with username and password in SSH
`vi /mnt/SDCARD/.tmp_update/config/scraper.json`

schmurtzm, 2023-11-10:
Should be resolved since Onion 4.2.0 final (and the PR #1110  indeed 😉 ).

closed 2023-11-10 by schmurtzm