OnionUI/Onion #1117: Wifi Issue

Hai1979, 2023-08-17:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

onion v4.2.0-rco

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

MM plus can not connect to wifi with 2.4 gb or 5 G wifi network and can not see in the wifi list 

### Steps to Reproduce

1. connect to MM plus wifti list 
2. type name of the route
3. type passwords
4. do not see it in the wifi list

### Provide any additional context or information that may help us investigate your issue

can not connect on mm plus

XK9274, 2023-08-17:
`or 5 G wifi network`

Hardware doesn't support 5gz networks - split your 2.4ghz off. 

Make sure your WiFi network doesn't contain special chars in either the SSID or the password.

Try using WPS if your router supports it.

If you continue to have issues:

`202303262339`

Try updating your firmware.

[See here](https://github.com/OnionUI/Onion/wiki/Frequently-Asked-Questions-%28FAQ%29#my-wifi-wont-connect-as-it-contains-special-characters-what-do-i-do)

spavon, 2023-08-22:
I'm experiencing a similar issue.

I've updated to the latest firmware: 202306092122 and the latest stable Onion build: 4.2.0-RC-snapshot-a5064207

I am able to use my phones hotspot and see other networks so I know that the wifi does work.

I've got a router with 2 APs configured to split the 5G and 2.4 bands.

2.4 AP is named with only alphabet characters. I've tried updating the password to remove special characters as well as removing the encryption altogether and still cannot see the network in a scan.

I've tried configuring the network to legacy mode but it hasn't helped.

I also tried using better-wifi-miyoo to manually add the network but connecting still fails.

Any help here would be much appreciated.

theswordsgame, 2023-08-26:
> `or 5 G wifi network`
>
> Hardware doesn't support 5gz networks - split your 2.4ghz off.
>
> Make sure your WiFi network doesn't contain special chars in either the SSID or the password.
>
> Try using WPS if [...]

Thing is I am having the same issue but this seems to be an onion specific issue cause on mainui and allium wifi connects fine

MassaKone, 2023-09-17:
Me on the other hand can connect my MM+ to my home WiFi whose password has special characters but I can’t connect the MM+ to my iPhone hotspot.

tGecko, 2023-09-17:
> Me on the other hand can connect my MM+ to my home WiFi whose password has special characters but I can’t connect the MM+ to my iPhone hotspot which SSID is basic and password too. 

Enable "maximize compatibility" in your iPhones hotspot settings. Without this setting, the hotspot is on 5Ghz only, which the MMP doesn't support.

MassaKone, 2023-09-17:
> Enable "maximize compatibility" in your iPhones hotspot settings. Without this setting, the hotspot is on 5Ghz only, which the MMP doesn't support.

I did that, I have “maximize compatibility” on and no bueno.

theswordsgame, 2023-09-18:
Strangely enough using onion desktop tools to connect to WiFi worked and it connects but using the default wifi setting in onion doesnt

theswordsgame, 2023-09-18:
I think this might be an issue with the built in wifi selector in onion as my other comments indicate

XK9274, 2023-09-24:
It's not technically Onions wifi selector, it's built into MainUI which is closed source. It does struggle with special characters. Try using: 

https://github.com/XK9274/better-wifi-miyoo

theswordsgame, 2023-09-24:
Sorry if I'm replying to this post too much but my wifi has no special characters it's all English letters and other os like allium, koriki, and mainui all instantly connect I am only having this issue with onion os

XK9274, 2023-09-24:
No it's fine.. It just seems odd that it could be Onion causing the problem when it uses the same firmware as the rest of the OS you listed but the difference is Onion uses Miyoos frontend where Allium and Koriki don't..

Out of curiosity can you try manually merging this file into your SD card: `.tmp_update/script/network/update_networking.sh`

https://github.com/OnionUI/Onion/blob/612fadf174cd972c99110b481564a225eef0228b/static/build/.tmp_update/script/network/update_networking.sh

After that you'll have to enable logging in tweaks then reproduce & attach the logs in `.tmp_update/logs` here

theswordsgame, 2023-09-24:
i did it also just curious what was the point of that .sh file?
[update_networking.log](https://github.com/OnionUI/Onion/files/12709112/update_networking.log)

XK9274, 2023-09-24:
It just fixes some syntax errors so we can see the actual output. 

go to tweaks > network > take a screenshot
flip some random toggles
close tweaks

then open simple terminal app with wifi enabled in simple terminal can you type:

ifconfig > ifconfig.log
ps > ps.log
dmesg > dmesg.log
wpa_cli list_networks >wpa_cli.log
iw dev > iw.log
lsmod > lsmod.log

and attach those (they'll be in the SD card root), the screenshot, aswell as runtime.log.

theswordsgame, 2023-10-05:
So for the delayed comment I noticed that the files include some personal details like IP's and name of wifi so just asking is it fine if I censor/delete those for privacy purposes

XK9274, 2023-10-05:
> Sorry for the delayed comment I noticed that the files include some personal details like IP's and name of wifi so just asking is it fine if I censor/delete those for privacy purposes 

The IP is probably your internal LAN IP so it's fine, but yes we won't need your WiFi SSID. 

Remove whatever you feel is personal info though :)

HollyCeuin, 2023-10-06:
I'm also having an issue with this - at random wifi will break and neither OnionOS or stock will discover _any_ networks. A reinstall fixed this yesterday, today it is not.

HollyCeuin, 2023-10-06:
Just gonna add to this, since I'm returning my MM+ for a new one, check your charging bricks. If they're not up to spec there's a pretty high chance it'll fry the wifi chip. That's what's happened to me, lol.

tGecko, 2023-10-06:
> Just gonna add to this, since I'm returning my MM+ for a new one, check your charging bricks. If they're not up to spec there's a pretty high chance it'll fry the wifi chip. That's what's happened to me, lol.

How can you be sure the charging brick fried the WiFi? What charging brick were you using?

HollyCeuin, 2023-10-06:
Wasn't using a brick, was using a power strip's USB ports, and because I've tested the ports now and they put out way too much voltage - other devices had issues on it (an AYN Odin's GPU driver resetting on screen touch while plugged into this) and I just hadn't put two and two together. The chip itself is no longer detected by the MM+, and while everything else works fine, there is a faint smell of burning emitting from it, so I'm putting 2 and 2 together.

tGecko, 2023-10-06:
How much voltage does the power strip supply to the usb ports?

apelete, 2023-10-08:
Hi @XK9274,

Following my proposal on the discord channel to help debug this issue, here are the logs I got on my MM+ :
[wpa_ping.log](https://github.com/OnionUI/Onion/files/12841063/wpa_ping.log)
[wpa_cli.log](https://github.com/OnionUI/Onion/files/12841064/wpa_cli.log)
[ps.log](https://github.com/OnionUI/Onion/files/12841065/ps.log)
[lsmod.log](https://github.com/OnionUI/Onion/files/12841066/lsmod.log)
[iw.log](https://github.com/OnionUI/Onion/files/12841067/iw.log)
[ifconfig.log](https://github.com/OnionUI/Onion/files/12841068/ifconfig.log)
[dmesg.log](https://github.com/OnionUI/Onion/files/12841069/dmesg.log)

FYI, my MM+ fails to connect to my homw wifi network:
- home network SSID : `Psyche`,
- home network encryption : `WPA/WPA2-Personal (mixed mode)`.

But it successfully connects to my smartphone access point:
- phone network SSID: `Pixel AS`
- phone network encryption: `WPA2-Personal`

Let me know if you need additional details.

apelete, 2023-10-08:
Also, here are screenshots of the Tweaks menu :
[image]
[image]

XK9274, 2023-10-08:
> Hi @XK9274,
>
> Following my proposal on the discord channel to help debug this issue, here are the logs I got on my MM+ : [wpa_ping.log](https://github.com/OnionUI/Onion/files/12841063/wpa_ping.log) [...]

What point were the logs taken? while connected to the non working network or the working one? 
What's your router? does it split 2.4ghz and 5ghz? Can you pull logs on the router?
Do you see the network in the list?
Wifi 6?
If your MMP is connecting to your phone it's not an issue with your MMP as such (well, a it's likely a compatibility issue) 

In terminal app you can try:

- wpa_cli (will drop you into a shell)
- interface wlan0
- add_network (will give you an id, eg 1)
- set_network 1 ssid "ssid"
- set_network 1 psk "pass"
- enable_network 1
- reassociate

and it should start the handshakes, you'll also see debug in here which will give you an idea of whats happening. 

you can also use

- scan
- wait 20 seconds or so
- scan_results

to see if it's appearing at all, but MainUI should list it if it appears in this scan which likely means it's not being picked up.

apelete, 2023-10-09:
> What point were the logs taken? while connected to the non working network or the working one?

Logs where taken while MP was not connected to any network.

> What's your router? does it split 2.4ghz and 5ghz?

Router is a WRT1900ACS flashed with OpenWRT LEDE.
Currently it operates in 2.4Ghz only, with 5Ghz disabled.

> Can you pull logs on the router? Do you see the network in the list? Wifi 6?

I looked at router logs, didn't see anything significant.
MMP does not even list the router's network (router's wifi does not even show on MMP).

> If your MMP is connecting to your phone it's not an issue with your MMP as such (well, a it's likely a compatibility issue)

Looks like a compatibility issue indeed, but MMP is my only device not able to see router's network (router wifi works just fine with my phone and all my computers).

> In terminal app you can try:
>
>     * wpa_cli (will drop you into a shell)
>
>     * interface wlan0
>
>     * add_network (will give you an id, eg 1)
>
>     * set_network 1 ssid "ssid"
>
>     * set_network 1 [...]

Tried those commands but it does not solve the issue : still unable to list or connect to the network.

> you can also use
> 
>     * scan
> 
>     * wait 20 seconds or so
> 
>     * scan_results
> 
> 
> to see if it's appearing at all, but MainUI should list it if it appears in this scan which likely means it's not being picked up.

No luck : even with those commands wifi network does not appear on MMP.

Please see screenshots attached for more details :
[image]
[image]
[image]

XK9274, 2023-10-10:
Thanks for all of this and for taking the time.. It's a curious issue tbh. All i can think is something is configured in the router that the MMP doesn't like. 

Could be the channel, could be the wifi tech, driver, chip on either device, who knows. Probably worth having a poke around in the router but i suspect you already have :).

Do you have a static channel set? Even though the devices say they support all 14 as per below maybe there's issues with certain ones? Idk - not having any issues here so hard to reproduce

My MMP wifi notes are here:

<details>

```
WiFi Adaptors detailed:

Wiphy phy1
	wiphy index: 1
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1962
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 150 Mbps
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
[... 122 lines cut ...]
		 * connect
		 * disconnect
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
	interface combinations are not supported
	Device supports SAE with AUTHENTICATE command
	Device supports scan flush.
	Supported extended features:
```
</details>

theswordsgame, 2023-11-05:
This probably isn't very important but the WiFi chooser in the settings of onion os connects to WiFi's without passwords instantly with no issue

kmcclry, 2024-01-31:
> @XK9274
> 
> TLDR: Some Routers/APs have "Mixed" or "Compatibility" mode that allows for 802.11b (WiFi v1), 802.11g (WiFi v3), and 802.11n (WiFi v4). MM+/OnionOS does not play nice with it. Telling the Router/AP to just use 802.11n (WiFi v4) resolved the issue.
> 
> I figured I should share some information I came across that may help people. It might want to be added in some of OnionOS Docs or WiFi Troubleshooting section.
> 
> I am using DD-WRT and only have the 2.4Ghz enabled. For some reason the MM+ would not always connect to the Wifi. Sometimes it was successful, sometimes it was unsuccessful (Home A). I knew 100% it was the fault of the router/access-point because this issue never occurred at my other residence, at all. The other place (Home B) uses a commercial grade UniFi AccessPoint and pfSense as the router. Every time I booted up the MM+ it would always connect and sync the time.
> 
> So why does Home B work and Home A not? During my troubleshooting I noticed that Home A would connect but not reliably. I downloaded and installed BetterWiFi to see if I can get more details at when/where the WiFi was failing. Turns out it was failing at the DHCP handshake for some odd reason. I have other devices on this network and I know that there was plenty of IP addresses to hand out.
> 
> Digging through the WiFi setup/config on the router/AP I realized that this setup was using a "Mixed" Network mode (some vendors might call it compatibility mode) were it allows for older devices and standards to be supported. I figured that maybe for some reason MM+/OnionOS was connecting periodically because it just got lucky and during the WiFi handshake/initialization with the Router/AP and chose the correct WiFi Standard version. As soon as I changed standard on the Router/AP to 802.11n (WiFi v4) and forced that to be the only option for all devices on my network, the MM+/OnionOS connected to the WiFi every time without issue.
> 
> Hope this information is helpful.

After googling around and trying to find fixes I wanted to say this worked for me as well. The moment I specified the network as 802.11n instead of "Mixed" I was able to get my MMP to connect. I haven't done many power cycles to see if it sticks (since some people have said their problem is intermittent I guess I could have just gotten lucky) but compared to how many times I've failed to get it to connect that fact that it worked right away when I did this I wanted others to have some more confirmation.

kmcclry, 2024-02-08:
Update. After two more start ups and shutdowns I'm having the issue again even though my router is set to 802.11n only.

I haven't owned a emulator hand held before this one so I'll try to get the logs figured out but it may take me awhile.

kmcclry, 2024-02-09:
@XK9274

I've done multiple attempts at troubleshooting this and taking the logs as you laid out earlier. The starting place was using the 802.11n network setup as the other user had suggested which worked one time for me and then never again after a few restarts.

The first set of logs is from starting up the device normally in the first network state and just running the logging you suggested.

The second set of logs are run immediately after manually trying to connect to the network using the wpa_cli suggestion you had. This was still on the 802.11n setup.

The third set of logs were run after I changed the 2.4ghz band to 802.11b/g instead of n. The MMP automatically connected when I turned it on so I actually have a connection in this one. After taking those logs I restarted again just to make sure it wasn't a one-time thing like before and I still have network connection. Not sure if this gives any insight that b/g only seems to work for me where n seems to only work for that other person or not, but I thought it maybe it helps.

As others have said. When using BetterWifi it was obvious that things were stopping at DHCP handshakes as that is where it errorred out when I tried connecting for those first two attempts. I'm not a network person so I have no idea what that implies, but I wanted to note that's what I was seeing.

I hope somewhere in the logs it may point to why one is working when the other isn't so this can get fixed for everyone!

[1 - MMP Wifi Log Files.zip](https://github.com/OnionUI/Onion/files/14227063/1.-.MMP.Wifi.Log.Files.zip)
[2 - MMP Wifi Log Files after another connection attempt.zip](https://github.com/OnionUI/Onion/files/14227064/2.-.MMP.Wifi.Log.Files.after.another.connection.attempt.zip)
[3 - MMP Wifi Log Files after b-g switch.zip](https://github.com/OnionUI/Onion/files/14227065/3.-.MMP.Wifi.Log.Files.after.b-g.switch.zip)

XK9274, 2024-02-27:
Thanks for all the logs and diags @kmcclry 

I'm not yet sure what we can do with this as i have never had the issue so can't really do many in-depth diags, not sure about the other guys in the team, Gecko did update his router lately and had no end of problems so reverted..

Looking at your logs (the second) it looks like some similar logs provided by other users, where the connection attempts but ultimately fails `RTW: link_timer_hdl:no beacon while connecting` and failing with an error code that means nothing unless you know the driver source. Your networks also go into a disabled state so it probably stops trying completely at some point.

It seems to me that it's more a router issue where it doesn't handle compatibility well with either older devices or, devices with this driver.

PS. 
As much as BetterWifi is useful it's ultimately not a diag tool, being stuck at DHCP means very little as it doesn't handle handshake failures. It's reached the point it's ready for DHCP but the connection has dropped.