OnionUI/Onion #1121: Turbo buttons not working on NES

judd43, 2023-08-20:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2 release candidate

### Miyoo Firmware Version

202306092122

### Provide a clear and concise description of the issue

Turbo/auto fire does not work for NES. I am on Onion 4.2 release candidate on Miyoo Mini Plus and have not changed any settings. Pressing X or Y in NES games doesn’t do anything. In the retro arch settings under port 1 controls, X is mapped to Turbo A and Y is mapped to Turbo B which is correct but they don’t work in game.

### Steps to Reproduce

Open any NES game using the default emulator and settings. Press X or Y to trigger turbo A and B. Notice how nothing happens. 

### Provide any additional context or information that may help us investigate your issue

That is all. Thank you!

Grayjks, 2023-08-25:
Try Nestopia core or install Nestopia in package, tubro is work!

ll6963, 2023-08-26:
Just make simple changes to get it done
quick menu >core options>input>turbo enble : none--player1
tubro is work!

closed 2023-08-26 by Aemiii91