OnionUI/Onion #1128: LAN Netplay issues with default SNES Core (Beetle Supafaust)

JoNilsson, 2023-08-24:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0-RC

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

When attempting a local LAN 2 player game.  All goes well until attempting to connect to the LAN host displayed after refreshing the list in the Netplay sub-menu.   

When the 2nd player attempts to connect, RetroArch displays a message

``` 100%:  No Core Found ```

Both devices are running identical everything.  

Anyone got any advise or ideas about what I could try to get this working?!   
Really craving a two player game of Donkey Kong Country with the wifey! 

Thanks in advance!

### Steps to Reproduce

Set-up a two player SNES Netplay session and attempt to connect on the 2nd player (Client) device.  

### Provide any additional context or information that may help us investigate your issue

I followed this guy's guide on Youtube:

https://youtu.be/QxrfblFXLTY?si=x-22XuPx8NJU-L6n&t=797

XK9274, 2023-08-24:
This is a known issue for supafaust, me and schmurtz did some experiments a few days ago and it's possible an RA issue and out of our hands.

This core works if you use easy netplay but there are some other issues with SNES in general.

Try opening GLO with your Y button and selecting easy netplay host, then on the client select easy netplay join. If you don't have these options in GLO and have the latest RC go to package manager in apps and enable them.

If you don't want to use easy netplay for whatever reason then check this spreadsheet.

https://docs.google.com/spreadsheets/u/0/d/1y1esG3HcNagPG8cy-ulZK_mhhUb93pqG5_XEtRRgKQ4/htmlview?pli=1#gid=0

You'll still need to change the core you've loaded the game with and the best way to do this is GLO.

JoNilsson, 2023-08-24:
> This is a known issue for supafaust, me and schmurtz did some experiments a few days ago and it's possible an RA issue and out of our hands.

Thanks so much for the reply.  I figured that it might be a RA issue.  I looked there, but there are so many open, that I will need to do more digging to see if there is already an issue regarding this raised in the RA repository.  If not, I will endeavor to open one.  

I'll try your solution utilizing easy net-play.  If I can figure it out, I'll try document the process and put in a PR to add the info to the SNES Wiki.

closed 2023-08-25 by JoNilsson