OnionUI/Onion #1133: miyoogamelist,xml ignored in expert section (for PC-98 at least)

pig-cop, 2023-08-27:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0-RC

### Miyoo Firmware Version

202306092122

### Provide a clear and concise description of the issue

If I add a core in the expert section that by default is missing the gamelist option in the _config.json_ and add it, the _miyoogamelist.xml_ file will be ignored. The games will be displayed including the boxart (if it's in the /Imgs/ subfolder and properly named) but just like they display by default when _miyoogamelist.xml_ is not used, so the filename will be used as a name in the list.

### Steps to Reproduce

This is what I have done:
- installed the PC-88 core (Neko Project II Kai)
- edited the _config.json_ file as this
`{"label":"Neko Project II Kai",`
`"icon":"/mnt/SDCARD/Icons/Default/rapp/nekop2kai.png",`
`"launch":"launch.sh",`
`"rompath":"../../Roms/PCNINETYEIGHT",`
`"imgpath":"../../Roms/PCNINETYEIGHT/Imgs",`
`"gamelist":"../../Roms/PCNINETYEIGHT/miyoogamelist.xml",`
`"useswap":1,`
`"shortname":0,`
`"hidebios":1,`
`"extlist":"hdi"`
`}`
- added a single game in the /Roms/PCNINETYEIGHT/ folder, in this case:
`Kono Yo no Hate de Koi o Utau Shoujo YU-NO (with Special Disk) (EN v1.01).hdi`
- added this _miyoogamelist.xml_ to the /Roms/PCNINETYEIGHT/ folder:
`<?xml version="1.0" encoding="utf-8" standalone="yes"?><gameList><game><path>./Kono Yo no Hate de Koi o Utau Shoujo YU-NO (with Special Disk) (EN v1.01).hdi</path><name>Kono Yo no Hate de Koi o Utau Shoujo YU-NO</name><image>./Imgs/Kono Yo no Hate de Koi o Utau Shoujo YU-NO (with Special Disk) (EN v1.01).png</image></game></gameList>`

After refreshing the list the game does appear but ignores the _miyoogamelist.xml_ file.
It does not use the name specified in the xml and ignores the specified image if the folder or name is different from the default one (/Imgs/<same_name_as_the _rom>.png).

### Provide any additional context or information that may help us investigate your issue

For testing if I'm doing something wrong with my _config.json_ and _miyoogamelist.xml_ I moved the core folder from
`/mnt/SDCARD/RApp/np2kai` to `/mnt/SDCARD/Emu/np2kai` and refreshed the list there, the miyoogamelist.xml is used as expected and all works like it should that way.
My guess is that maybe it's a bug with the expert cores, or with PC-98 in particular?

deframe, 2023-09-02:
I had the same problem with the expert Hatari core. Not only was the `miyoogamelist.xml` file ignored, it actually appeared in the ROMs list, as did the `Imgs` folder.

pig-cop, 2023-09-02:
Yeah happens to the PC-98 core too.
It's because the default `config.json` generated in the `/RApp/<core_name>` folder has a null value for the extension list and I guess it is assumed in this case to show all extensions.
You can edit the line with `"extlist"` and define the correct extensions for that core.

Leemann07, 2024-02-05:
I had the same problem with the expert dosbox Pure core.