OnionUI/Onion #1136: Netplay doesn't work on MM+

mmatviyiv, 2023-08-28:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0-RC

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Hi folks,
I know Retroarch-related bugs should be addressed to their repo, but looks like it's something related to Onion's network component.

So, I have Retroarch (same version, same core, same from file) running on Mac, PS Vita, and MM+.
State autosave settings are disabled as Wiki describes. All devices in the local network.
I've managed to pair Mac and Vita, Netplay worked just perfectly there. But unfortunately, none of them can connect to MM+.

I've tried Relay mode and direct connection by IP, the outcome is the same - it connects and then gets disconnected in a second without any errors. The same result if MM+ is connecting as a client and the Host.

I checked Retroarch log file, nothing useful there, unfortunately:
```
[INFO] [Config]: Looking for config in: "/mnt/SDCARD/RetroArch//.retroarch/retroarch.cfg".
[INFO] === Build =======================================
[INFO] CPU Model Name: ARMv7 Processor rev 5 (v7l)

[INFO] Capabilities: NEON VFPv3 VFPv4
[INFO] Built: Jul 27 2023
[INFO] Version: 1.15.0
[INFO] =================================================
[INFO] [Input]: Found input driver: "sdl_dingux".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: 320x240.
[INFO] [CPU]: Path not found: ./cpuclock.txt
[INFO] [Joypad]: Found joypad driver: "sdl_dingux".
[INFO] [Video]: Found display server: "null".
[INFO] [OSS]: Using audioserver.
[INFO] [Display]: Found display driver: "null".
[INFO] [Playlist]: Loading history file: "/mnt/SDCARD/Saves/CurrentProfile/lists/content_history.lpl".
[INFO] [Playlist]: Loading history file: "/mnt/SDCARD/Saves/CurrentProfile/lists/content_music_history.lpl".
[INFO] [Playlist]: Loading favorites file: "/mnt/SDCARD/Saves/CurrentProfile/lists/content_favorites.lpl".
[INFO] [Core]: Using content: "/mnt/SDCARD/Roms/MD/Contra - Hard Corps (USA, Korea).zip".
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] [Autoconf]: Dingux Gamepad disconnected from port 1.
[INFO] [Video]: Does not have enough samples for monitor refresh rate estimation. Requires to run for at least 4096 frames.
```

To me it seems like Onion just drops the connection at the very last moment for some reason.
Would be much appreciated for your thoughts and help.

### Steps to Reproduce

Just connect with Onion device to any other with Netplay.

### Provide any additional context or information that may help us investigate your issue

Provided above

mmatviyiv, 2023-08-29:
Another strange observation that could be related - SSH, FTP (probably other protocols as well, hadn't tested them) had stopped working unexpectedly today on my MM+, Onion was just dropping connections despite having been working just fine yesterday.
Reboot, wifi reconnection, re-enabling them in tweaks were not helping.
Worked again after I reinstalled the tweaks app. Really odd.

XK9274, 2023-09-04:
It's likely to do with save states. You can usually fix this by disabling some settings in retroarch.cfg (below pasted from schmurtz in discord here: https://discord.com/channels/529983248114122762/1087760354278387722/1102962957563985971)

```
playing with RA options a little : 
==============================

always working (but losing automatic savestates in Onion is not a good idea) : 
savestate_auto_save = "false"

Working (probably the best solution for now, implemented in the current 4.2.0 beta  https://github.com/OnionUI/Onion/commit/6a1cdcf5c0f1e62a4dde89392036cdf2ca823e1f): 
savestate_auto_save = "true"
savestate_file_compression = "true"
savestate_thumbnail_enable = "false"

Freeze (previous 4.2.0 config) :
savestate_auto_save = "true"
savestate_file_compression = "false"
savestate_thumbnail_enable = "true"

Freeze :
savestate_auto_save = "true"
savestate_file_compression = "false"
savestate_thumbnail_enable = "false"

crash : 
savestate_auto_save = "true"
savestate_file_compression = "true"
savestate_thumbnail_enable = "true"
```

Which is being addressed by this PR:
https://github.com/OnionUI/Onion/pull/1140

mmatviyiv, 2023-09-05:
thanks for the reply @XK9274 
state autosave params options were the first things I tried. Unfortunately, none of these combinations worked for me, it just drops the connection despite it being disabled or enabled, so must something else

XK9274, 2023-09-05:
> thanks for the reply @XK9274 
> state autosave params options were the first things I tried. Unfortunately, none of these combinations worked for me, it just drops the connection despite it being disabled or enabled, so must something else

That's an odd one then, alright:

- When you changed the settings did you modify the .cfg or do it through RA?
- What core are you using?
- What ROM? (CRC is fine)
- Does easy netplay work for you? (If you have 2 MMP)
- Have you tried enabling retroarch logging to see if it gives any more info?

mmatviyiv, 2023-09-05:
> * When you changed the settings did you modify the .cfg or do it through RA?
> * What core are you using?
> * What ROM? (CRC is fine)
> * Does easy netplay work for you? (If you have 2 MMP)

1) Through RA, then saving the current config to .cfg via menu
2) The latest PicoDrive 1.99, tried different ARMv7 builds along with the Onion's precompiled one
3) Tested with multiple Sega MD roms, for example `Contra - Hard Corps (USA, Korea)`, CRC `C579F45E`
4) Ordered the second device a few days ago, it may take a while since it's an international shipment

mmatviyiv, 2023-09-05:
Oh, seems I found the disconnection reason, finally have managed to get Netplay logs and here's the error:
```
[INFO] [Netplay] Connected to: "Anonymous"
[ERROR] [Netplay] Netplay state load with an unexpected save state size.
[INFO] [Netplay] Netplay disconnected
```

That's quite interesting since all auto-save and state compressions are disabled on both Host and Client, so sizes must match.

Would be much appreciated if anyone could test a cross-platform connection and try to reproduce that. Wounder whether it's a global MM issue or not.

XK9274, 2023-09-12:
I've seen this error a few times while testing something else and i'm pretty sure it was with the same game. Have you tried adjusting any settings on the other platforms? I've had PC -> MM+ Netplay working in the past but i ended up changing some save state settings (compression etc) on both platforms IIRC.