OnionUI/Onion #1137: MMP - Onion V4.2.0-RC - 'About Device' settings option freezes the device

rowest, 2023-08-29:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion V4.2.0-RC

### Miyoo Firmware Version

Can't check rn because of the mentioned issue

### Provide a clear and concise description of the issue

The 'About Device' settings option freezes the device - buttons don't respond, game switcher button vibrates but does nothing as well. Only holding the power button works (restart the device).
I have 2 Miyoo Mini Plus devices and the issue replicates on both of them. 

### Steps to Reproduce

Turn on Miyoo Mini Plus, go to Settings -> About device - tap "A"

### Provide any additional context or information that may help us investigate your issue

Sometimes the screen goes black after pressing some buttons, sometimes it just stays on the settings options screen and freezes

pig-cop, 2023-08-29:
I reported this last week: https://github.com/OnionUI/MainUI-issues/issues/17

If you wait a bit over a minute it eventually unfreezes and show the info.
I sure hope it will get fixed.

specialmixer, 2023-09-02:
Same here.
Tested it with 2 different Miyoo Mini Plus, with FW 0609 and 0628 and 4 different SD cards
The same problem also occurs when clicking OTA Update.
After waiting 90 seconds it works.
But the checkdisk prompt is not displayed on OTA.
Once you have waited 90 seconds, it works immediately until after a restart.
I've already been in touch the developer on FB.
Unfortunately there is no solution so far.

XK9274, 2023-09-04:
Duplicate of https://github.com/OnionUI/MainUI-issues/issues/17 but leaving open until decided whether it's an Onion issue or a MainUI issue.

I'm unable to reproduce this on 3 MMP with different levels of Firmware (0505, 0609, 0628), any SD card and across various versions of Onion.

There's a couple of hw level things to note that the about device section checks that may be causing this:

- Serial number from `/config/riu_r`
- Firmware version from uboot env using `/etc/fw_printenv`
- Used/total storage on SD, C function `statvfs` is called for this and checks `/mnt/SDCARD`

If you're also getting this in OTA though it's possible its an SD card thing. 

- Are all your SD cards formatted in the same way? If so how? 
- When you tested 4 SD cards did you clone them or was it just a fresh install of Onion?
- Do you get this behaviour in stock with an SD not containing Onion inserted?

The first function to run in OTA is:

`check_available_space`

Which makes a call to check available space on a mounted area, can you run this via telnet/ssh/uart and see if you get any delays:

```
mount_point=$(mount | grep -m 1 '/mnt/SDCARD' | awk '{print $1}') 
available_space=$(df -m $mount_point | awk 'NR==2{print $4}')
echo $mount_point && echo $available_space
```

It should be instant.

pig-cop, 2023-09-04:
I can confirm that running that via ssh took about the same time to run, about one minute.
But only the first time, it behaves exactly like the reported issue.
My SD is a 128GB Samsung EVO formatted in FAT32 (about 10% full).
If I insert the the unbranded SD the MMP came with, on stock there's no issue.

specialmixer, 2023-09-04:
Connected via ssh takes the same time.
The 4 SD cards are always a fresh install. There are 3x Samsung Evo Pro Plus 128GB and 1x Samsung Evo Plus 128GB
They are formatted with fat32format (allocation unit size 512).
If I start the Miyoo  without an SD card, I can call up the device information without freezing.

XK9274, 2023-09-04:
Loading without an SD will cause the above to be skipped and show unmounted anyway so it's looking like an issue with the card.

Does it show the correct size when you insert it in windows then view the size through right click properties?

Try repairing it in windows when you insert it also to remove any dirty flags

Do you have any other cards to test onion with?

specialmixer, 2023-09-04:
Can you just copy the data from one card to another or do you always have to do a fresh installation?

XK9274, 2023-09-04:
Yeah man just sling it over, aslong as its formatted FAT32.

pig-cop, 2023-09-04:
When you said about the size I noticed that it was wrong on Windows, so I tried copying my files on a brand new pre-formatted Lenovo microsd I had laying around here and it worked without the issue.
I re-formatted my problematic Samsung EVO (also brand new) with 'fat32format' and the size was wrong again.
So I tried formatting it with Rufus (non bootable, Large Fat32, no extended label and icon), the size was finally right.
Copied my files back, put the microsd in and now it works!
I guess there's something going on between 'fat32format' and Samsung EVO microsd cards...

XK9274, 2023-09-04:
Awesome, glad it's sorted & thanks for reporting back! Could you close your issue down over on MainUI if you're happy it's sorted? 😊

@rowest 
@specialmixer 

Can you check the above comment by @pig-cop and see if formatting with a different tool such as Rufus fixes for you?

specialmixer, 2023-09-05:
Problem also solved.
When I format my 128GB card with fat32format, it only shows a size of 53.5GB in Windows.
I formatted the cards with MiniTool Partition Wizard (allocation unit size fat32 = 64k). 
Now everything works without problems.

Thanks @pig-cop & @XK9274 
for the great cooperation.
This way the problem could be found and fixed quickly.

rowest, 2023-11-12:
thanks for the info @pig-cop

closed 2023-11-12 by rowest