OnionUI/Onion #1161: optimized version of ffplay

schmurtzm, 2023-09-25:


XK9274, 2023-09-30:
Have built this.

In it's current state not fit for purpose.

Launches via an app but nothing on screen, playing sound but possibly at incorrect sample rate and no joystick input. 

It does initialise mi_vdec though, so maybe an issue with screen initialisation as it's done via a header file.

Will rebuild at a later date to diagnose screen 👍

<details>

welcome to test ssplayer!
[MI_PANEL_Init][332]LCD environment is Invalid
mmplayer audio_dev = 0
ssd20x_getpanel_wh = [640 480]
try playing /mnt/SDCARD/Media/Videos/FFplay controls.mp4 ...
mmplayer video_rotate = 0
mmplayer video_only = 0
mmplayer video_ratio = 1
mmplayer enable_scaler = 0
mmplayer resolution = 8294400
mmplayer play_mode = 1
### enter mm_player_open
ssplayer library version: git_commit.d76ef58 build_time.20230925
set out width/height = [640 480]
ffmpeg_version..git_commit:d76ef58..build_time:20230925
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x276b0] [libavformat/utils.c 3749]set video max_resolution [8294400]
avio buffer size = 32768, probesize = 5000000
avformat demuxer name : mov,mp4,m4a,3gp,3g2,mj2
frame_mbs_only_flag = 1
max_bytes_per_pic_denom = 0
frame_cropping_flag = 0
conformance_window_flag = 0
video codec w/h = [720 480]
audio idx: 1, video idx: 0
start time : 0.000, total time of input file : 69.419
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/mnt/SDCARD/Media/Videos/FFplay controls.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 512
    compatible_brands: mp42iso2avc1mp41
    creation_time   : 2023-01-06T17:53:07.000000Z
    encoder         : HandBrake 1.5.1 2022011000
  Duration: 00:01:09.42, start: 0.000000, bitrate: 399 kb/s
    Stream #0:0(und): Video: h264 (avc1 / 0x31637661), yuv420p(tv, bt709), 720x480 [SAR 8:9 DAR 4:3], 75 kb/s, 25 fps, 25 tbr, 90k tbn, 180k tbc (default)
    Metadata:
      creation_time   : 2023-01-06T17:53:07.000000Z
      handler_name    : VideoHandler
    Stream #0:1(eng): Audio: aac (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 157 kb/s (default)
    Metadata:
      creation_time   : 2023-01-06T17:53:07.000000Z
      handler_name    : Stereo
    Stream #0:2(eng): Audio: aac (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 157 kb/s
    Metadata:
      creation_time   : 2023-01-06T17:53:07.000000Z
      handler_name    : Stereo
avformat_seek_file success!
open_demux successful
open video codec: ssh264
origin w/h=[640 480], screen w/h=[640 480], sar4:3 w/h=[640 480], sar16:9 w/h=[640 360]
set vdec out w/h = [640 480]
scaler src w/h = [640 480], dst x/y/w/h = [0 0 640 480]
[ssh264 @ 0x75fa0] video prefix format : yuv420p.
[ssh264 @ 0x75fa0] h264 has b frames : 2
(ss_h264_vdec_init 631)exec function pass
(ss_h264_vdec_init 634)exec function pass
(ss_h264_vdec_init 653)exec function pass
(ss_h264_vdec_init 654)exec function pass
[ssh264 @ 0x75fa0] set h264 esbuf size = [2097152]
(ss_h264_vdec_init 660)exec function pass
[ssh264 @ 0x75fa0] ssh264 video w/h = [720,480], vdec output w/h = [640,480]
[ssh264 @ 0x75fa0] ss_h264_vdec_init successful!
type of nal : 7. len : 32, data : 67,64,0,1f
type of nal : 8. len : 6, data : 68,e9,78,23
ssh264 find valid sps, pps nal header!
[ssh264 @ 0x75fa0] ss_h264_decode_init successful!
codec width: 720,height: 480
display w/h = [640 480], src w/h = [640 480]!
video time base : 0.011111 ms.
fps : 0.040, frame rate num : 25. frame rate den : 1.
get in video decode thread!
open_video successful
open audio codec: aac
audio channel layout[3]
video fps time : 0.040
video_playing_thread in
avcodec_flush_buffers for video!
 _MI_AO_OpenVqeLib: success
 _MI_AO_OpenSrcLib: success
 _MI_AO_OpenG711Lib: success
 _MI_AO_OpenG726Lib: success
mm_audio_init dev[0] done!
open_audio successful
leave mm_player_open ###
get file duration time [69.419]
player display first picture [728263]us
avcodec_flush_buffers for audio!

</details>

XK9274, 2023-11-30:
https://github.com/aaron201912/ffmpeg/issues/3