OnionUI/Onion #1162: allow arcade/Mame 2003+ to quit with menu button

schmurtzm, 2023-09-25:
It's not possible to exit arcade games by simply pressing menu button.

Few time ago I've made this video : https://www.youtube.com/watch?v=Ode06KGi3QQ to explain to users the specialties of this platform. You can see it the first time you run an arcade game in Onion (it's also available in the video player).

The problem comes from Retroarch : it doesn't react as it should to the linux "kill" (sigterm) command. Normally the kill command should exit Retroarch properly but for many arcade games it just not work : Retroarch is not exited.

So I have created some issues to Retroarch repo : 
For Retroarch : https://github.com/libretro/mame2003-plus-libretro/issues/1623
And this one for Mame2003+ : https://github.com/libretro/RetroArch/issues/15717

The interesting fact is that configuring a hotkey to exit Retroarch works properly.... So the workaround is to drive Retroarch from the [Network command line API](https://docs.libretro.com/development/retroarch/network-control-interface/) : this QUIT command exits arcade properly and it makes the save state. So we have a workaround 🎉

 Many other network commands [are not documented](https://github.com/libretro/RetroArch/blob/041ae3010332da1155d1ac6d6368dd05d2c3003e/command.h#L448) and it will be interesting to investigate on the possibilities that it gives.

Example to drive retroarch from windows :  `sfk199.exe udpsend 192.168.0.203 55355 QUIT` 
(You'll find [sfk199 here](https://sourceforge.net/projects/swissfileknife/files/latest/download?source=files)).

It will require some little modifications in Onion to implement this :  We justs need to have `HAVE_COMMAND = 1` in our RA build  and `network_cmd_enable = "true"` in the retroarch.cfg and then we need to update keymon to manage this new API instead of kill/sigterm.

mmc41, 2023-10-21:
Does this also fix that shutdown button is ignored as well during arcade games ?

schmurtzm, 2023-11-05:
> Does this also fix that shutdown button is ignored as well during arcade games ?

it should yes ;)

JakiShen, 2024-01-14:
Add Gamelist.xml file generator please

seanstardev, 2024-01-31:
I love the OP's idea and description for a solution here. Great out of the box thinking :)

I've noticed that some arcade games still don't exit when 'Menu' button is pressed. Has anyone else observed this, or is it possible I've mucked up my settings? Happy to provide specific problem games if that helps :)

mahoneyt944, 2024-08-08:
https://github.com/libretro/RetroArch/pull/16854 should fix this

JamesIV4, 2024-12-03:
When can we hope for this fix to be in an Onion update? I switched to using Mame2003-Plus from Mame-2003-Extreme and found out about this issue that way. I figured support is better for Mame2003-Plus because it's not considered an Expert core.

Would it be possible for me to update retroarch manually to get this fix sooner?

I'll probably wait, just thinking out loud.