OnionUI/Onion #1167: Add Drastic / NDS emulation

schmurtzm, 2023-09-25:


schmurtzm, 2023-10-20:
Done in https://github.com/OnionUI/Onion/tree/FEAT-NDS-integration

Still require some work (when it's possible) :  

- [ ] move saves
- [ ] move bios
- [ ] Onion background menu logo
- [ ] documentation
- [ ] manage auto save states / Game Switcher & auto resume at boot ?

WingofaGriffin, 2024-02-10:
What is the process for keeping this up to date with the third party core? This isn't using submodules, so it is just a manual update? I say this as I notice there was an update upstream that doesn't look to be reflected.

LuisGustavo78, 2024-02-13:
> Qual é o processo para manter isso atualizado com o núcleo de terceiros? Isso não está usando submódulos, então é apenas uma atualização manual? Digo isso porque percebo que houve uma atualização upstream que não parece ser refletida.

you can see it in commits, there you can see what the onion team is working on, yesterday Sicrotefo updated Drastic to 1.8, and since steward-fu is working in collaboration with the onion team, they will always keep it updated

schmurtzm, 2024-02-13:
Hi, cores and emulators are not updated automatically in Onion, that's just not possible because there are a lot of breaking changes and performances regressions on updates. 

I had a similar question about Dosbox recently so I copy paste my answer here 😉  : 

About the date of each core, we have an automated build of most cores however we don't use it to update Onion systematically : the core selection is mostly manual and must stay like this for many of the reasons I'll mention next.
I already know this objective for some users to have all "up to date" but clearly it's not adapted to Onion. In Onion we select our cores with a lot of care : we test it, we compare it and sometimes we modify or improve it before integrate it in Onion. It's no coincidence that most other distributions for the Miyoo Mini use our cores ;) . 
But what is interesting is that being up to date  on the cores doesn't mean to have the best stability, the best performances and the best features. In fact, the opposite is often true. For example if you build the current Daphne you’ll not have the sound. If you build the current scummvm you will have no image. if you build the last snes9x 2002 it will be slower ...
Also updating a core can break the save states which are heavily used in Onion and it could create a disparity of the cores which is not ideal for Netplay.
However there are some exceptions : for example we work with the gpSP author and we update the core regularly. This collaboration has allowed to modify the core deeply without breaking all the save states of the Onion community. 

What could interest you is [this file](https://docs.google.com/spreadsheets/d/1_QQIuV3LTJP5Yf7fa3fEYTfcMq8m0w_3Az93_yv_7KY/edit?usp=sharing) (not updated recently with last Onion releases) that I use to follow the cores state of Onion.
I will be happy to have your thinking on what should be updated and why. Then it's not a problem to include it in a next Onion release after having checked that all the parameters are good  ;)

LuisGustavo78, 2024-02-13:
Thanks for clarifying, I didn't know that either!