OnionUI/Onion #1190: Pokemon hotspot - Client 'disconnects' but then syncs saves

kjm1988, 2023-10-13:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.2.0-rc-snapshop-a5064207

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Easy Pokémon Netplay (local hotspot) disconnects a client after 1-2mins of playing Pokémon TCG but then successfully syncs saves on closing RetroArch.

This issue doesn't occur when using our home WiFi network and manually connecting to each other but the client's save game isn't synced as it is when using the Easy Pokémon Netplay option via Hotspot.

Same model, same onion, same firmware, same ROM and we're sat next to each other.

Any suggestions please?

### Steps to Reproduce

1. Boot both Miyoo's and press Y on Pokemon Care Game (USA) on both.
2. Select 'Pokemon Trade/Battle: Host Hotspot' on Miyoo 1 wait 5 seconds and then select 'Pokemon Trade/Battle: Join Hotspot' on Miyoo 2.
3. Play for a couple of minutes and then the client disconnects (sat next to each other). Game desyncs.
4. Close RetroArch on Miyoo 1, it waits then on the client to close RetroArch before attempting to sync the save game.
5. Close RetroArch on Miyoo 2, save games successfully sync, implying both devices are still connected.

### Provide any additional context or information that may help us investigate your issue

I'm happy to run through any tests and/or provide logs if someone could point me in the right direction. I will test on other GBC Pokemon ROMs over the coming days.

I'm happy to use my local WiFi as a work around rather than the hotspot Pokemon script if someone could tell me where to manually put the client saves on the host Miyoo/what the easy Pokemon script does as this has been absolutely solid in my testing.

XK9274, 2023-10-16:
There's some changes coming to this feature in 4.2 release - we can have a look into it then if the problem still exists 👍.

Generally though it's a connection issue, RetroArch can be quite weird with netplay if there's a bad connection. 

Funnily enough aswell, when I was testing some Netplay features the closer my MMP were together the worse latency was. Other users also had this behaviour.

kjm1988, 2023-10-21:
@XK9274 
We're still encountering the same issue in 4.2.2. The problem is only with 'Easy Netplay - Pokemon Trade/Battle', all other forms of netplay work well across 8 & 16bit emulators for us.

When hosting with hotspotting from Miyoo #1 the client disconnects after 1-2 minutes gameplay but then the saves sync successfully and the script ends gracefully, when hosting from Miyoo #2 the host fails to receive any save games, while Miyoo #1 (now the client) hangs on 'Syncing our save files with the host'. 

Is it possible to have a Pokemon Trade/battle script that copies the Gambatte saves to TBG Dual and then from the client to the host but using normal WiFi instead of hotspotting? It's the hotspotting that seems to be the issue.

Gameplay works well when manually hosting and joining Pokemon games via RetroArch in TGB Dual (aka using local WiFi instead of hotspotting) but player 2 is missing all their Pokemon as their save isn't copied across.

I can't really code but please let me know if there's anything I can do to help troubleshoot.

kjm1988, 2023-11-03:
I've tested this in 4.2.3 and everything seems to be working perfectly now. Thank you!

I've opened another ticket as a few Netplay options are essentially duplicated but only one of them works but I'm closing this ticket.

Thank you for your work on OnionOs, it's amazing!

kjm1988, 2023-11-05:
After working flawlessly for one time only play Pokémon TCG after upgrading I'm seeing the same issue. Client 2 disconnects after 1-2 minutes playing via Hotspot, playing using normal WiFi is fine but obviously missing saves. Hosting on the other Miyoo now works but also disconnects. I have 2 more mini's on the way so hopefully I can rule out a hardware issue.

techfooninja, 2024-06-03:
I'm seeing this as well. I'm on Onion OS 4.3.0 with firmware 202306282128. I've found that shutting down both devices through the Settings menu (not just pressing and holding the power button) and turning the devices back on gets us to where we can have a stable connection. But it seems that the next connection is in a weird state, and we have to power down and restart again.

XK9274, 2026-01-21:
You're correct on the complexity of ENP, it was time sensitive but not only that a pain to maintain as it had a ton of duplicated logic and functions with a lot of moving parts. 

A lot has changed in EN since this issue, and i've got another PR up for some more refactors here https://github.com/OnionUI/Onion/pull/1891 - It may not fix the disconnect issue, and as you mention normal WIFI is a lot more stable I'm starting to think maybe it's something to do with the hotspot being unstable at a hardware level - Can't see what else it could be as the script doesn't really do anything when RA is running.

I've turned off power save (both ints) in this PR to see if that improves things at all & it may be that we neeed to look at how wlan1 is configured when we bring the hotspot up against it to see if there's any more stability to be pulled out of it. There's also been a lot of changes to RA since we first added these, some to netplay itself so could also contribute to a more stable experience. 

The end goal would be a proper binary instead of the shell scripts and TCP or at the least UDP comms between the devices, aswell as proper sync across the whole process.. Worth it for MMP if the HW can't handle it? I'm not sure.

Going to assume you may have moved on from this after over a year of the issue being sat so I'm going to close it, but feel free to reopen it again if you're still getting the problem.