OnionUI/Onion #1212: Retroachievements Not Working

OldGameDad, 2023-10-19:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.1

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

I've an issue encountered that is not fixed with this update, and in consistent across my 2 Miyoo Mini devices, but NOT my other wifi enabled retro handhelds through retroarch, is being unable to properly 'use' in to Retro Achievements. I am correctly logged into Retroachievements, and when the system ATTEMPTS (usually it does not do this) to communicate with retroachievements, it will say "could not communicate with [retroachievements.org](https://retroachievements.org/)" Most of the time, it will say "not logged in" in the Retroarch settings, even though I am properly logged in to retroacheivements in all areas, and I have attempted all corrective fixes through searches. I am also on an approved firmware for 4.2.

As a note, wifi causes major slowdown to launching and closing a game, loading and saving a save state, which are directly tied to Wifi being on.

### Steps to Reproduce

Have wifi on. Ensure your retroachievement login information is entered in both areas accurately. Go to launch a game that has retroachievements. Encounter no retro achievements message or one that states "could not communicate with [retroachievements.org](https://retroachievements.org/)"

### Provide any additional context or information that may help us investigate your issue

[image]
[image]

OldGameDad, 2023-11-05:
This issue still occurs in 4.2.3.

schmurtzm, 2023-11-10:
Looks like a network problem, like a DNS resolution problem from your Miyoo Mini. Are you able to use other network services on Onion (like Scraper or Standard Netplay for example) ?

OldGameDad, 2023-12-02:
My apologies for the delay on this, I updated to the 2023 06 09 2122 firmware and the issue is more concrete now, that under the Quick Menu > Achievements, I get "Not logged in" only now. 

I am able to easily scrape, OTA, and other network services at all times.

Yin26, 2024-12-04:
I'm encountering this problem as well. GameSwitcher runs extremely slow with retroachievements turned on. Load states and save states are barely useable. Overlays take as much time to load as the time it takes for retroachievements to connect/not connect. It would sometimes work but 9 times out of 10 it is riddled with this slowdown. I'm not new in emulation I have an Odin 2 but I got an MM+ thinking I would be able to enjoy the classics in this form factor with my retroachievements but it seems this issue went past my radar. I hope it gets fixed soon.