OnionUI/Onion #1219: Tweaks : restore default setting with start button

schmurtzm, 2023-10-20:
The idea is to be able to restore the default setting for an option in Tweaks. This behavior would be the same as Retroarch.
Useful for example in "`Button shortcuts`" to restore the default value quickly. 

An alternative could be to indicate the default value directly in the text of the option, for example : 
`MainUI: Y button`       `GLO (Default)`

... but we'd certainly run out of room to do this

theUnBurn, 2023-10-30:
Hey! I like this feature and wouldn't mind taking this one; would that be alright with you @schmurtzm ?

schmurtzm, 2023-10-30:
We haven't started to work on it so any participation is very welcome 😉
The 2 important stuffs to determine are probably where/how to store the default value and how to display it in the UI.

theUnBurn, 2023-10-30:
Actually, whats the difference between this function and the "Reset Settings" feature in Tweaks?

XK9274, 2024-01-13:
> Actually, whats the difference between this function and the "Reset Settings" feature in Tweaks?

I guess it's so you can reset a single setting. Reset settings will wipe the complete config directory:

https://github.com/OnionUI/Onion/blob/c988db28fe2cb50f4bc81f74535938fe755088a4/src/tweaks/reset.h#L73C1-L73C1

I still think it's a decent idea, the default value could be show in the info when you hit select on an item.. but as for storing the defaults idk.