OnionUI/Onion #1221: activity tracker : acutalizing boxart after the game has been added to the BD

schmurtzm, 2023-10-20:
Currently adding a box art (with Scraper or else) doesn't actualize the game cover in Activity Tracker.

This is due to the fact that Activity Tracker stores the image path in a db instead of generating the image path dynamically.

Aemiii91, 2023-10-20:
I think we should just get the box art image dynamically (by checking if the png exist) - instead of having to keep a db up-to-date

theUnBurn, 2023-12-06:
@schmurtzm I made a fix for this issue here: https://github.com/OnionUI/Onion/pull/1305

Would love feedback! Particularly around on if we should update the db to no longer have an image path (and, if so, how to do that).

JakiShen, 2024-01-14:
Add a Gamelist.xml generator please

Prozi, 2024-02-11:
> @schmurtzm I made a fix for this issue here: #1305
> 
> Would love feedback! Particularly around on if we should update the db to no longer have an image path (and, if so, how to do that).

I love the mr https://github.com/OnionUI/Onion/pull/1305 please merge it to next onion dev release please

Prozi, 2024-02-12:
@Aemiii91 how can I support MR #1305

Prozi, 2024-02-19:
> Would love feedback! Particularly around on if we should update the db to no longer have an image path (and, if so, how to do that).

I wouldn't remove the field in db or update it, this way:

if someone installs newer onionos, has some activities in db, then decides to downgrade, his db will still work with older onion (backwards compatibility)

schmurtzm, 2024-02-20:
Hi, I have a problem with this PR... Curiously I'm not able to triggers the initial problem . Hum...😅
Quickly tested the actual Activity Tracker and I don't see any problem : changing the cover of a game impacts Activity Tracker too. I don't remember in which case the fact that Activity Tracker keeps the path of the image is a problem anymore. 😶

Prozi, 2024-02-21:
> Hi, I have a problem with this PR... Curiously I'm not able to triggers the initial problem . Hum...😅 Quickly tested the actual Activity Tracker and I don't see any problem : changing the cover of a [...]

in that case @schmurtzm at least please reopen https://github.com/OnionUI/Onion/issues/1419 because that was a bit different (missing images on games, dont update, dont show, nothing, in activity tracker, but they are scraped prior, more details in linked issue)

Prozi, 2024-02-21:
the first example I recall is Tony Hawk's Pro Skater 4 on PS

Prozi, 2024-02-25:
@Aemiii91 any progress on this one if this is a duplicate of 1441 as you said there?

Prozi, 2024-02-25:
@schmurtzm did you try this on any PSX games like I said in my issue? did you try this with THPS4?

also, see:

https://github.com/OnionUI/Onion/assets/3052357/4ebe536c-aa32-4fcb-8f52-5d324136828a

Prozi, 2024-02-27:
@schmurtzm @Aemiii91 because of your lack of cooperation on the issue I've decided to download the sqlite file and check it with online SQL tool, it shows that images for few games are empty records

[image]

I will edit it manually and this is what I recommend anyone having problems with Activity Tracker to do in the future:

1. download `/Volumes/MIYOO/Saves/CurrentProfile/play_activity/play_activity_db.sqlite`
2. visit https://extendsclass.com/sqlite-browser.html
3. upload the downloaded sqlite file
4. enter in the textarea `SELECT * FROM rom;`
5. choose RUN button
6. fix any errors by running `UPDATE rom SET image_path="__PATH_TO_IMAGE__" WHERE id=0;` change `0` to row id and `__PATH_TO_IMAGE__` with path to actual image starting with `/mnt/SDCARD/Roms`

good luck

-----

by the way some of you two said you can't reproduce it, how did you actually try? what did you do? did you try to change the image path to `__A$$HOLE__` and then update it through scraper? or what?

XK9274, 2024-02-27:
Thanks for the feedback, Lets keep it constructive.