OnionUI/Onion #1227: Fatal - Shutdown+gameswitcher not working (v222)

mmc41, 2023-10-21:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

V2.2.2

### Miyoo Firmware Version

20230609

### Provide a clear and concise description of the issue

After installing Onion V2.2.2 everything seems to work at first but if I try an an arcade game (f.x. 1942) the "game switcher" button and shutdown button stops working. If I press them a sound is played and then nothing happens. Can't switch out of the game and if I disconnect the battery the arcade game launches automatically again and I am stuck again (The anti-feature of auto-running the last game is particularly annoying in this case).

### Steps to Reproduce

Install onion V2.2.2 on Myioo mini plus and activate arcade
Add mame arcade roms with 1942 game
Run 1942 game
Try press gameswitcher - nothing happens but a sound
Try shutdown -  nothing happens but a sound

### Provide any additional context or information that may help us investigate your issue

This also happened with an earlier beta and earlier firmware.

Aemiii91, 2023-10-21:
It's a well known issue, and we're not able to fix it at the moment. You have to exit arcade games through the RetroArch menu (MENU+SELECT)

mmc41, 2023-10-21:
0) There was no such problem with the RetroArch that came with the device (without Onion).
1) It would be nice to document such fatal known issues! There was no warning in the release notes or the wiki.
2) How can I avoid arcade games to rerun on start so I am constantly stuck?
3) How do shutdown the device when stuck in a arcade game?

tGecko, 2023-10-21:
1. You are shown a video when you first start an arcade game that explains how to exit, how to save and load save states. 
2. Hold the menu button while booting.
3. Open the RetroArch menu using MENU+SELECT and select `Quit RetroArch`

closed 2023-10-21 by Aemiii91